k3ng/k3ng_cw_keyer #12: E24C1024.h not found

mweyland, 2015-08-26:
When trying to compile the current git version for a Due, compilation aborts with the error:

k3ng_keyer.ino:386:24: fatal error: E24C1024.h: No such file or directory

This is with FEATURE_EEPROM_E24C1024 set in keyer_features_and_options.h and HARDWARE_ARDUINO_DUE  set in keyer_hardware.h

I downloaded the ARM build environment from inside the Arduino 1.6.5 IDE. This file does not seem to be part of the stdlib or anything. Probably this is just a simple case of a forgotten 'git add E24C1024.h'

Regards and thanks
Matt

mweyland, 2015-08-26:
Follow-up: Maybe not a simple case of a forgotten git add after all. This header file seems to be mentioned here: http://playground.arduino.cc/Code/I2CEEPROM24C1024 and the 'lastest' zip file linked in on that page, and available at https://code.google.com/p/arduino-at24c1024/downloads/list seems to make it compile after a manual update to the current Arduino API. I probably just missed the spot in your docs where you tell us where to get the current version from that library. Sorry for the noise.

k3ng, 2015-08-28:
I currently don't include this library in the git, but I guess I could.
I'll go ahead and add it on the next upload.  73

mweyland, 2015-08-28:
Awesome! If there is a maintained version out there, it would suffice to link that in my opinion. But unless I missed something, the way it is now is that it's not quite clear where to get it from and the last maintained version I could find was quite outdated.

73 and thank you, Matt

k3ng, 2015-08-29:
I have added the files to the git, and also fixed a bug with Arduino Due compilation using this library.  Thanks for the heads up!

73
Goody
K3NG

closed 2015-08-29 by k3ng