k3ng/k3ng_cw_keyer #19: Command Mode: Add Callsign Practice

flickerfly, 2016-02-02:
Feature Request: Add the ability to start callsign practice mode from command mode. I propose using C as the command to trigger it.

_I'll take a whack at this at some point unless someone beats me to it which is cool._

If I understand what I've read correctly, currently the only way to start up call sign practice mode is to have the CLI feature turned on to trigger it. It seems like it would be useful to be able to trigger this from command mode; especially for those using the smaller arduinos since CLI takes up a lot of program space.

I believe this needs to be implemented in the command_mode() function in k3ng_keyer.ino around line 4731. I'm looking at implementing basically a copy of this when I have a bit more time to look at it and figure out how the case statements work.

        #ifdef FEATURE_ALPHABET_SEND_PRACTICE
          case 111:
            send_dit(AUTOMATIC_SENDING);
            command_alphabet_send_practice(); // S - Alphabet Send Practice
            stay_in_command_mode = 0;
            break;
        #endif  //FEATURE_ALPHABET_SEND_PRACTICE

I think the function to call is us_callsign_practice(), but I have a good bit of research before being sure how this should work.

k3ng, 2016-02-08:
Hi Josiah.  If you haven't attempted to code this yet, I will attempt, hopefully this evening.

flickerfly, 2016-02-08:
Neat, I haven't touched it yet and probably wouldn't think of it this week. Thanks!

k3ng, 2016-02-08:
I was about to implement this, however I just realized/remembered that the callsign practice requires the CLI.  You use it to type in the callsigns you hear.

k3ng, 2016-02-08:
Sorry, didn't mean to close.  Let me know if you want to close or pursue an alternate callsign practice that doesn't need the CLI.

flickerfly, 2016-02-09:
Oh,  cool! Could that be made optional?

k3ng, 2016-02-09:
It could, but it would require some sort of alternate way for the user to input the callsign they hear.  Perhaps send the same callsign back?  (Could be done, but a good bit of work I think.)

flickerfly, 2016-02-09:
Could it simply be put on the screen requiring manual verification? The
software then would have no feedback. Maybe this should be broken into a
whole other feature, just sharing the code that creates the calls.  Sounds
like some substantial changes.

gsansoucie, 2018-01-28:
Based on what you did with the Echo mode, this would be perfect to:
a) Require FEATURE_DISPLAY
b) For each Callsign
c) Clear the screen
d) Display the expected callsign on the screen
e) User attempts to key the callsign
f) Verification of keyed in text similar to the method implemented in the Echo 5 mode
g) Repeat

I saw the commented out code in the command_progressive_5_char_echo_practice() which is where I thought this was going (then I realized it was copied from the CLI method).

I finally finished my keyer today, the enclosure arrived this afternoon while we were out (longest project ever waiting on parts).  I might work on this locally and pull some of the extensive CLI CW Training stuff into the Command Mode interface.

gsansoucie, 2018-01-30:
Anthony, after I got the WINKEY Interface feature working (all I did was uncomment it and clean up an #ifdef) I am able to use my k3ng keyer with Iambic Master (https://sites.google.com/site/korkowp1/iambic-master) which is going to suit me for sending practice.  I might toy around with this Callsign send practice in the background as it would work in stand alone mode (vs needing to be hooked up to a Windows PC for Iambic Master).

k3ng, 2018-02-02:
Hi Glen.  Thanks for the note.  You mentioned you had to clean up an
#ifdef?  Did you have to make a modification to get the Arduino code to
work with Iambic Master?

73
Goody
K3NG

gsansoucie, 2018-02-02:
The compiler complained about the “defined” keyword/macro that was uncovered by defining the FEATURE_WINKEY.   That was it.

-=Glen=-

k3ng, 2018-02-02:
What change did you make?

gsansoucie, 2018-02-03:
Anthony,

Sorry it took so long, I finally sat back down on my MBP.

It was a simple misplaced #ifdef  and defined(), can’t have both:

@@ -9491,7 +9492,8 @@ void service_winkey(byte action) {
             #ifdef DEBUG_WINKEY
               debug_serial_port->println("service_winkey: WINKEY_ADMIN_COMMAND 0x01");
             #endif //DEBUG_WINKEY          
-            #ifdef defined(__AVR__) //#ifndef ARDUINO_SAM_DUE
+            // 30 Jan 2018 GAS Fix
+            #if defined(__AVR__) //#ifndef ARDUINO_SAM_DUE
               asm volatile ("jmp 0"); /*wdt_enable(WDTO_30MS); while(1) {};*/ 
             #else
               setup();

k3ng, 2018-02-06:
Thanks.  Fixed in version 2018.02.05.01.  73 Goody

gsansoucie, 2018-02-06:
Oh good, I was actually going to branch again to put in the fix and issue a pull request (I’ve never really used git before, I’m coming around from cvs/rcs/svn). My clone is currently full of CW Command line send code that you don’t want.

-=Glen=-