k3ng/k3ng_cw_keyer #29: TFT Support

k3ng, 2016-12-06:


4x6hp, 2018-01-07:
Will be NEXTION support?

k3ng, 2018-01-13:
I haven't looked at these units.

VK2EFL, 2018-08-04:
Anthony,
Some time ago I purchased online a low cost 3.5" TFT display 320x480  (~US$8) which had an SD card slot too.
It was kind of neat but it needed to be mated with an Arduino Uno like a daughter board. Doing that made for a nice integrated package but it took up all the pins. The only way to interface with the Uno and the TFT display was through the USB port.
I proceeded to install libraries to drive the display and compiled some code to emulate what a K3NG keyer might produce. I messed about with colours and fonts. Using a different library it was possible to use bit mapped fonts that did not get 'blocky' when using larger fonts. Graphical stuff is all nice but it used more memory on the Uno.
But the clincher here is that there was no way I could get any K3NG code in there with all the other stuff that dealt just with the display. In any case there were no pins available for interfacing with anything external.

I believe the Nextion displays are kind of similar where they contain a processor that handles all the display functions and the external system communicates with the display using serial commands (or something similar). This off-loads all the display driving code and processing to the display. I think the Nextion displays are also about 3 to 4 times the price of the cheap 3.5" display I got.

Fred
VK2EFL

IndyRick, 2021-01-01:
I'm thinking of taking a run at using a Nextion for the K3NG keyer. I've used a Nextion for my "own" (borrowed, and credited, open source code where needed - WW9JD Arduino uMatic2 Memory Keyer). Because of all the memory needed for Nextion button assignments, I used a 2560 Mega. My keyer used very few pins, and had lots of memory left over.

tulipman, 2021-01-01:
Hi Rick

A Nextion display on a k3ng keyer would be awesome!

Are there any piccies available of the Nextion display in action on a K3NG keyer?

Best rgds and happy new year!!

De Robbert / TA2IX

IndyRick, 2021-01-01:
Thanks. I couldn't find any examples of someone interfacing a Nextion to a K3NG keyer. It may easily be more than I can handle, but the trying should prove interesting.

closed 2026-10-23 by k3ng