k3ng/k3ng_cw_keyer #31: WinKey Params not compiling

Idunbehad, 2017-01-05:
Latest version software (1-5-17):

k3ng_keyer:6876: error: 'WINKEY_LOAD_SETTINGS_PARM_1_COMMAND' was not declared in this scope

      case WINKEY_LOAD_SETTINGS_PARM_2_COMMAND:

This continues for all 15 Params.  Also other issues with Winkey Emulation too numerous to list here.

Thanks for any help.  Been trying to run down the problem for 2 days with no luck.

k3ng, 2017-01-05:
Could you post all the features and options you're attempting to compile
with?  I can't recreate the error, but it may be specific to a certain
combination of features and options.  Thanks

Idunbehad, 2017-01-07:
These are the features/options I have uncommented:

FEATURE_COMMAND_BUTTONS
FEATURE_COMMAND_LINE_INTERFACE
FEATURE_MEMORIES
FEATURE_WINKEY_EMULATION
FEATURE_POTENTIOMETER
FEATURE_LCD_SAINSMART_I2C
FEATURE_CW _DECODER
OPTION_WINKEY_DISCARD_BYTES_AT_STARTUP
OPTION_WINKEY_2_SUPPORT

All other FEATURE/OPTIONS are as originally defined in KEYER FEATURES AND OPTIONS. h

Thanks--W0XR

Idunbehad, 2017-01-08:
Hi Anthony:I found the problem late this afternoon.  I am very new with Arduino so I goofed up somewhere. 

Apparently the libraries and header files were not in the right place to compile and some of them may have been corrupted.
Anyway, everything is compiling OK now.
I have a red face and apologize for taking up your time when you very well good have been helping someone who really needed it.
Idun(Jack)--W0XR

k3ng, 2017-01-08:
No worries!  Thanks for following up.  73 and Happy New Year!

Idunbehad, 2017-01-09:
Oboy, here we go again.Got the keyer to working fantastically.  

However:When I try to compile with Sainsmart 2004 LCD display activated, I keep getting;
k3ng_keyer:1577: error: 'lcd' was not declared in this scope

        lcd.clear();
I have the headers #include <Wire.h>
#include <LiquidCrystal_I2C.h>
#include <LiquidCrystal.h> 

Lost again:  W0XR

k3ng, 2017-01-10:
Could you post the entire compilation log, in verbose mode?

closed 2017-01-28 by Idunbehad