k3ng/k3ng_cw_keyer #33: Keypad added

Idunbehad, 2017-01-31:
Sir Anthony:
I have added a 4x4 keypad to the code and eliminated the memory resistor ladder and pushbutton switches.  The only button that remains active is the COMMAND button. 

10 memories are recognized.  Using a 4x4 keypad, there are 6 keys that are not used, but these could be configured to do  a multitude of tasks.  Such as using the "C" key to enter COMMAND MODE.

The implementation is using an Arduino Mega 2560.

The code will also recognize a 3x4 keypad.

Are you interested in this code?

Jack-W0XR

k3ng, 2017-02-01:
Hi Jack, I'm interested!  Can you do a pull request so I can view the changes and merge?

73
Goody
K3NG

Idunbehad, 2017-02-01:
Hi Anthony:No problem, but it may not be today.
I have to try to get it all together as there are additions to some of the header files also.
The code is quite simple.  It does require the use of "keypad.h" .
Will be working on this tomorrow.
73--Jack--W0XR

Idunbehad, 2017-02-01:
Keypad Code:

Include Keypad.h:    https://github.com/kernd/Arduino-Libraries/blob/master/Keypad/Keypad.h

Add just above Variables & Stuff:
#if defined(FEATURE_4x4_KEYPAD)||defined (FEATURE_3x4_KEYPAD)
  #include <Keypad.h>
#endif

Add to keyer_pin_settings.h:
#ifdef FEATURE_4x4_KEYPAD
  #define Row3 33
  #define Row2 32
  #define Row1 31
  #define Row0 30
  #define Col3 37
  #define Col2 36
  #define Col1 35
  #define Col0 34
  #endif

 #ifdef FEATURE_3x4_KEYPAD
  #define Row3 33
  #define Row2 32
  #define Row1 31
  #define Row0 30
  #define Col2 36
  #define Col1 35
  #define Col0 34
 #endif

Add to keyer_features_and_options.h:
 #define FEATURE_4x4_KEYPAD                  //Use this for 4x4 Keypad 
//  #define FEATURE_3x4_KEYPAD               //Use this for a 3x4 Keypad

Add to keyer_settings:
#define ROWS 4                            //KeyPad Rows
#define COLS 4                            //keypad Columns
//KeyPad Columns--Comment out if using 3x4 keypad
//#define COLS 3    //Uncomment for 3x4 keypad
#define mem1 0                            //Memory numbers for Keypad: Actual memory numbers start with "0"
#define mem2 1
#define mem3 2
#define mem4 3
#define mem5 4
#define mem6 5
#define mem7 6
#define mem8 7
#define mem9 8
#define mem10 9
#define mem11 10
#define mem12 11

Add to keyer_debug.h
// #define DEBUG_KEYPAD_SERIAL

Add this routine under //Subroutines----------------------------------------------------------
void KeyPad_use(){
  
// Define the Keymap for 4x4 matrix keypad

#ifdef FEATURE_4x4_KEYPAD
char keys[ROWS][COLS] = {
  {'1','2','3','A'},
  {'4','5','6','B'},
  {'7','8','9','C'},
  {'#','0','*','D'}
};
#endif

// Define the Keymap for 3x4 matrix keypad
#ifdef FEATURE_3x4_KEYPAD
char keys[ROWS][COLS] = {
{'1','2','3'},
{'4','5','6'},
{'7','8','9'},
{'#','0','*'}
};
#endif

// Setup for 4x4 matrix keypad
#ifdef FEATURE_4x4_KEYPAD
    byte rowPins[ROWS] = {Row3,Row2,Row1,Row0};  //Arduino Mega Pins: 33,32,31,30---Keypad Pins 5,6,7,8
    byte colPins[COLS] = {Col3,Col2,Col1,Col0};  //Arduino Mega Pins: 3735.34.33---Keypad Pins 1,2,3,4
#endif

#ifdef FEATURE_3x4_KEYPAD
    byte rowPins [ROWS] = {Row3,Row2,Row1,Row0};  //Arduino Mega Pins:  33.32.31.30--Keypad Pins 4,6,7,2
    byte colPins [COLS] = {Col2,Col1,Col0};       //Arduino Mega Pins:  36,35,34-----Keypad Pins 5,1,3
#endif

// Create the Keypad
Keypad kpd= Keypad( makeKeymap(keys), rowPins, colPins, ROWS, COLS );

#ifdef DEBUG_KEYPAD_SERIAL
   Serial.begin(9600);   //Use this to echo back to computer. 
#endif

  char key=kpd.getKey();
if(key)                      // Check for a valid key.
//   Serial.println(key);    //Used for checking only.  Not used in normal operation.
  {
    switch (key)
    {
      case '1':
        play_memory(mem1);    //MEMORY 1
      break;
      case '2':
        play_memory(mem2);   //MEMORY 2
      break;
      case '3':
         play_memory(mem3);  //MEMORY 3
      break;
      case '4':
         play_memory(mem4);  //MEMORY 4
      break;
      case '5':
         play_memory(mem5);  //MEMORY 5
      break;
      case '6':
        play_memory(mem6);   //MEMORY 6 
      break;
      case '7':
      play_memory(mem7);     //MEMORY 7
      break;
      case '8':
         play_memory(mem8);  //MEMORY 8
      break;
      case '9':
         play_memory(mem9);  //MEMORY 9
      break;
      case '0':
        play_memory(mem10);  //MEMORY 10
      break;
     case '#':
        beep_boop(); 
     break;
   case '*':
     beep_boop();
   break;
    case 'A':
       beep_boop();
    break;
    case 'B':
      beep_boop();
    break;
    case 'C':
    command_mode();
      //beep_boop();
    break;
    case 'D':
      beep_boop();
    break;
    }
    }
}
// end KeyPad_use;         

COMMENTS:
The resistor ladder and pushbutton switches have been removed.

The 10k resistor  and pushbutton switch for COMMAND have been retained.  

When the program is in COMMAND mode, it is not looking at the keypad, so the button has to be pushed to get out of that mode.  The button will still put it into COMMAND MODE.  I tried to find the proper place to put the keypad routine into the COMMAND interpreter, but haven't found it so far!

I did use the “C” on the keypad to go into COMMAND MODE, but it takes the pushbutton to get it back out or “x” from the key paddles, but if in COMMAND/TUNE mode, it still takes the pushbutton to get back out.

Note that I have not tested this will all available modes of operation.  I do not have a  PS2 keyboard attached or using a CW keyboard except through the Arduino Serial Monitor.

I also added a MEMORY MACRO:   ' \a' :  This clears the LCD display after a MEMORY has been sent.  I am still working on clearing the LCD of the information that has already been sent.  When a memory is triggered, the LCD displays all that has been sent previously, including any “hand-keyed” information.

NOTE;  I am using a 4x20 LCD.  The code had to be modified as follows:
#if defined(FEATURE_LCD_SAINSMART_I2C)
   #define I2C_ADDR      0x27
   #define BACKLIGHT_PIN 3
   #define En_pin        2
   #define Rw_pin        1
   #define Rs_pin        0
   #define D4_pin        4
   #define D5_pin        5
   #define D6_pin        6
   #define D7_pin        7
   LiquidCrystal_I2C lcd(I2C_ADDR,En_pin,Rw_pin,Rs_pin,D4_pin,D5_pin,D6_pin,D7_pin, BACKLIGHT_PIN, POSITIVE);  
 // LiquidCrystal_I2C lcd(0x27,20,4);  The line had to be commented out as the address 0x27 was defined in two places.
#endif //FEATURE_SAINSMART_I2C_LCD    

Have fun:  Hope it works for you.

73--Jack--W0XR

closed 2017-02-01 by Idunbehad

k3ng, 2017-03-31:
Added feature in version 2017.03.30.01

Idunbenhad, 2017-03-31:
Hi Goody:I just got back on, so Thanks for your efforts.  I will download and test the code shortly.
Thanks again.  It's no fun being stuck in a hotel, that's for sure!!!
73 for now
Jack--W0XR