k3ng/k3ng_cw_keyer #35: Keyboard.h missing

telerainbow, 2017-02-16:
Keyboard.h is not included in the project.

```
#if defined(FEATURE_CW_COMPUTER_KEYBOARD) 
  #include <Keyboard.h>
#endif //defined(FEATURE_CW_COMPUTER_KEYBOARD)
```
[link](https://github.com/k3ng/k3ng_cw_keyer/blob/master/k3ng_keyer/k3ng_keyer.ino#L799)

Maybe i missed some instructions but I can't find the file anywhere. 

73 and thank you for this great piece of software :)

telerainbow, 2017-02-16:
This issue was caused by an old version of Arduino IDE (1.6.4) available via fedora 23 repositories. I installed the newest version from the website. I had to use [this fix](https://github.com/arduino/Arduino/issues/5783), replacing the included avrdude with the system one because of dependency on libtinfo.so.5/libncurses.so.5 (fedora installs version 6).

It's working now, i'm leaving this here for others having similar problems.

best 73

closed 2017-02-16 by telerainbow

k3ng, 2017-02-17:
Thanks so much for figuring this out.  I became aware of the issue recently
but lacked the time to pursue a fix.  Thanks for the help and 73!  Goody
K3NG