k3ng/k3ng_cw_keyer #37: Warnings

Zinci, 2017-11-16:
I get 3 warnings while compiling.
1.) 14688:5: warning: "/*" within comment [-Wcomment]
2.) :5297:27: warning: statement has no effect
        sending_mode == AUTOMATIC_SENDING_INTERRUPTED;
3.)  5185:19: warning: unused variable 'endtime'
        unsigned long endtime = micros() + long(element_length*lengths*1000) + long(additional_time_ms*1000);

Anyone gets this?
I can easily just delete the first one but the rest is much more difficult. 
Also, they are just warnings. It does program the board.

m-chichikalov, 2017-11-16:
Don't worry.  In my opinion this is not a big deal...

Zinci, 2017-11-16:
So you get them as well? I thought I am doing something wrong. It does upload to board, no problems.
Did you by any chance remove the comment from "#define FEATURE_MEMORIES" in keyer_feautres_and_options.h? It does come with another warning at the line 12567:1. 
I was able to remove this warning by inserting the "return 0;" just before that, line 12566. I don't know if I broke the #ifdef FEATURE_MEMORIES.

k3ng, 2019-12-07:
These errors were fixed recently.  73 Goody K3NG

closed 2019-12-07 by k3ng