k3ng/k3ng_cw_keyer #38: Sequencer

4x6hp, 2017-12-18:
Adding sequencer routine with TAIL and LEAD delays. Declare pins for PTT PA and PTT TRX.
This option will be useful for HF PA. First switch TX on PA side, second switch TX on TRX side.

k3ng, 2018-01-07:
So the sequence would be PTT PA active, PTT TRX active, ptt_tx active, then tx_key_line active?

4x6hp, 2018-01-07:
For foot switch connected to pin defined for **tx_key_line** (active fro key down/TX on)
For WINKEY emulation PTT signal come from USB port or COM port connected to the keyer to pin defined for **ptt_tx**

k3ng, 2018-01-13:
I'm not quite understand what you are saying.  Could you list out the events like

1. this happens
2. this happens
3. etc....

Thanks and 73

4x6hp, 2018-01-14:
1. tx_key_line is active
2. after configured delay PTT PA is active
3. after configured delay PTT TRX is active
4. Transmitting CW 
5. tx_key_line is inactive
6. after configured delay PTT TRX is inactive
7. after configured delay PTT PA is inactive

I mean to add something similar to PC5M device here is documentation: [https://kvgog.home.xs4all.nl/projects/resources/tx_rxv2/tx_rx_sequencerv2.pdf](https://kvgog.home.xs4all.nl/projects/resources/tx_rxv2/tx_rx_sequencerv2.pdf)

f6fvy, 2018-03-03:
While you're at it, you could add a LNA PTT output and follow the sequence described in the MKII manual (http://www.microham.com/Downloads/MK2_English_Manual.pdf - pages 35/36).

k3ng, 2018-03-04:
> Downloads/MK2_English_Manual.pdf - pages 35/36).

I don't see that in the manual on pages 35 and 36.  Could you elaborate?

f6fvy, 2018-03-04:
Oops. Pages 32/33 (PTT tab section)

k3ng, 2018-03-04:
Here's how I'm thinking of implementing:

Pin definitions:

sequencer_1_pin
sequencer_2_pin
sequencer_3_pin
sequencer_4_pin
sequencer_5_pin

Time settings:

ptt_active_to_sequencer_1_active_time_ms
ptt_active_to_sequencer_2_active_time_ms
ptt_active_to_sequencer_3_active_time_ms
ptt_active_to_sequencer_4_active_time_ms
ptt_active_to_sequencer_5_active_time_ms

ptt_inactive_to_sequencer_1_inactive_time_ms
ptt_inactive_to_sequencer_2_inactive_time_ms
ptt_inactive_to_sequencer_3_inactive_time_ms
ptt_inactive_to_sequencer_4_inactive_time_ms
ptt_inactive_to_sequencer_5_inactive_time_ms

So you could independently set the times independently for each.

To mimic the timing in the diagram here
https://kvgog.home.xs4all.nl/projects/resources/tx_rxv2/tx_rx_sequencerv2.pdf
on page 2, you could configure like this:

#define sequencer_1_pin 20
#define sequencer_2_pin 21
#define sequencer_3_pin 22

#define ptt_active_to_sequencer_1_active_time_ms 0
#define ptt_active_to_sequencer_2_active_time_ms 50
#define ptt_active_to_sequencer_3_active_time_ms 100

#define ptt_inactive_to_sequencer_1_inactive_time_ms 100
#define ptt_inactive_to_sequencer_2_inactive_time_ms 50
#define ptt_inactive_to_sequencer_3_inactive_time_ms 0

How does this look?

I'm also considering having the timing setting configurable in the CLI and
stored in EEPROM rather than hard coded in defines at compile time.

4x6hp, 2018-03-04:
Looks very fine. Exact the sequencer!

f6fvy, 2018-03-04:
Sounds great. If you intend implementing setting these timings with the CLI interface, may I suggest to implement setting the PTT tail and PTT hang timings (currently set as #define) as well, because they're part of the "PTT sequence" as well.

k3ng, 2018-03-04:
Good idea.  Once I figure out the commands for sequencer timing, it's not
that much more work to add PTT timing.

Depending on how my day goes, I may work on this a bit today.

f6fvy, 2018-03-04:
BTW, I don't think there's provision for a PTT In (e.g. footswitch) input. If there was one, the sequencer outputs you described could be used in any mode, not only CW.

k3ng, 2018-03-04:
So a PTT In input would basically activate PTT and the sequencer pins?

f6fvy, 2018-03-04:
Absolutely. So, if your LNA, your amp, etc are wired to the keyer, you can use its sequencer section in any mode.

k3ng, 2018-03-04:
That should be doable.

k3ng, 2018-03-11:
I just about have this feature completed.  I'm doing testing and I'll need
to write up a wiki page on it.

k3ng, 2018-03-11:
Just uploaded code to Github with FEATURE_SEQUENCER.  Documentation is here:

https://github.com/k3ng/k3ng_cw_keyer/wiki/383-Feature:-Sequencer

Please test and let me know your results.

73
Goody
K3NG

k3ng, 2018-03-13:
Anyone do any testing?

4x6hp, 2018-03-13:
Tommorow I will let you know

closed 2018-03-20 by k3ng