k3ng/k3ng_cw_keyer #39: CW side tone frequency

technoplusnl, 2018-01-26:
Hi,

Got a simple version running on a Mini-Pro.
I seem to be unable to change the side tone frequency, it stays at 600Hz whatever I fill in at 
#define initial_sidetone_freq 500 
in keyer_settings.h
If I change #define initial_speed_wpm to 12 in the keyer-settings.h, that will work.

Thanks!
73 de Albert, PA3GUK

gsansoucie, 2018-01-26:
I can attest to the fact that the CMD mode method of changing the side tone works on the UNO, I keep accidentally selecting that mode ;-)

k3ng, 2018-01-27:
Albert, initial_sidetone_freq sets the sidetone frequency at first run of
the code; it writes it to EEPROM.  You could use command mode or the CLI to
change the sidetone frequency at runtime, or if you do a "factory reset" by
squeezing both paddles and powering up which would write the value to
EEPROM.

technoplusnl, 2018-01-27:
Thanks, squeezing both paddles during startup does the trick for the initial side tone frequency!
In CLI mode I can change mode with "\a" and "\b" but when I type "\f0800" I get ERROR.
Please see attached log file.
Thanks!

Thanks

[error.txt](https://github.com/k3ng/k3ng_cw_keyer/files/1670087/error.txt)

k3ng, 2018-01-28:
I'm not able to reproduce that issue.  What version of code are you using
and what features and options are you compiling with?

technoplusnl, 2018-01-29:
Thanks. I'm using version 2018.01.14.01 on an Arduino Uno with few options switched on.
Please find the config files attached.

Albert

[keyer_features_and_options.zip](https://github.com/k3ng/k3ng_cw_keyer/files/1674212/keyer_features_and_options.zip)

gsansoucie, 2018-01-30:
Anthony,

Glen here (gsansoucie/n1xf).  I wanted to let you know that I am able to use Iambic Master with the FEATURE_WINKEY_EMULATION
option.

I added an FTDI module to my k3ng project (This one: ttps://www.amazon.com/gp/product/B00N4MCS1A) and it basically just worked.

Unfortunately I didn’t know about this option when I was building my keyer and I still have to figure a way to mount the little board, but that is a different issue (solved with JB Weld and Duct Tape no doubt).

Thanks for making this project available.

-=Glen=-

n1xf

k3ng, 2018-02-02:
Albert, are you using the Arduino IDE serial monitor and you have the line
terminator set to "Newline" rather than "Carriage Return" or "Both NL & CR"?

technoplusnl, 2018-02-03:
Thanks! setting it to "Both NL & CR" did the trick!

k3ng, 2018-02-03:
Good deal.  73

Goody
K3NG

closed 2018-02-03 by k3ng