k3ng/k3ng_cw_keyer #43: command mode speed 255 after download

g4fre, 2018-03-03:
building the 25 feb load on a teensy 3.2
I have set" #define initial_command_mode_speed_wpm 20 // "factory default" command mode speed setting"
when i download it gets set to 255

\S
Iambic A
Buffers: Dit On Dah On
WPM: 15
Command Mode WPM: 255
Sidetone: On 1500 hz
Dah to dit: 3.00
Weighting: 50
Serial Number: 1
Potentiometer WPM: 25 (Activated) Range: 13 - 35 WPM
Wordspace: 7
TX: 1
Quiet Paddle Interrupt: 52
Mill Mode Off
Memory 1: CQ CQ CQ DE G4FRE
Memory 2: G4FRE
Memory 3: WW2R
Memory 4: MW2I
Memory 5:{empty}
Memory 6:{empty}
Memory 7:{empty}
Memory 8:{empty}
Memory 9:{empty}
Memory 10:{empty}
Memory 11:{empty}
Memory 12:{empty}

if i do a factory reset I get the same result
I cannot find a command to set the command mode speed, and i cannot key at 255

the build errors are:-

k3ng_keyer: In function 'void loop_element_lengths(float, float, int)':
k3ng_keyer:5564: warning: statement has no effect 
              sending_mode == AUTOMATIC_SENDING_INTERRUPTED;

                           ^

k3ng_keyer:5459: warning: unused variable 'endtime' 
     unsigned long endtime = micros() + long(element_length*lengths*1000) + long(additional_time_ms*1000);

                   ^

k3ng_keyer: In function 'void service_paddle_echo()':
k3ng_keyer:11035: warning: unused variable 'character_to_send' 
   byte character_to_send = 0;

        ^

k3ng_keyer:11036: warning: unused variable 'no_space' 
   static byte no_space = 0;

               ^

k3ng_keyer: In function 'void serial_tune_command(usb_serial_class*)':
k3ng_keyer:11683: warning: variable 'incoming' set but not used 
   byte incoming;

        ^

k3ng_keyer: In function 'byte play_memory(byte)':
k3ng_keyer:14597: warning: control reaches end of non-void function 
 }

 ^

k3ng_keyer: In function 'void serial_program_memory(usb_serial_class*)':
k3ng_keyer:13805: warning: 'memory_number' may be used uninitialized in this function 
           EEPROM.write((memory_start(memory_number-1)+memory_index),255);

                                     ^

i think the last one is because i have set memories to 4 as I only want 4 buttons \s lists 10?

Dave
g4fre

k3ng, 2018-03-03:
If you change various settings, wait maybe 40 seconds, then reboot, are the
settings you changed preserved?

g4fre, 2018-03-04:
Yes, only this one parameter so far I have found is corrupted

 

I did just try loading a new teensy 3.2 and it does the same

 

Dave

 

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Virus-free.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

g4fre, 2018-03-04:
One other thing I noticed when trying to debug the issue is that if you uncomment out “#define DEBUG_MEMORYCHECK” in keyer_debug.h,  you get the error “'SP' was not declared in this scope” and can go no further

 

Dave

 

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Virus-free.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

k3ng, 2018-03-04:
> auth/ACJAxaqjW41nxqnFaCiTi3Mv_yKFOWPZks5taylTgaJpZM4SbGH4> .

When you do the factory reset, do you hear a beep-boop-beep-boop-beep-boop?

k3ng, 2018-03-04:
Try the latest code I just uploaded to Github.  I found a minor issue that
may have been causing this.

g4fre, 2018-03-04:
I Just saw the notification and tried it. That has fixed the problem with command speed

 

What us mill mode? It seems to default “on” but I haven’t found out what it does or how to permanently disable it!

 

Thanks for the fix. I can now go ahead and make the pcb for the miniature winkey 

 

dave

 

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Virus-free.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

k3ng, 2018-03-04:
> ACJAxSLgajGT9fWnMkBKiWA37p2aDcDrks5tbCMHgaJpZM4SbGH4> .

It's a feature that is somewhat in development / still evolving.  It's
described a bit here:

https://groups.io/g/radioartisan/message/7915?p=,,,20,0,0,0::relevance,,mill,20,2,0,12388497

/@ should turn it off.

g4fre, 2018-03-04:
Thanks for the info. Yes I had found that command, Don’t think I have a need for it. I hope there will  be a line to disable it in the features file

 

I just did a hack to get a “winkeyer mode active LED “ indicator

 

Dave

 

https://groups.io/g/radioartisan/message/7915?p=,,,20,0,0,0::relevance,,mill,20,2,0,12388497

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Virus-free.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

k3ng, 2018-03-07:
> ,20,0,0,0::relevance,,mill,20,2,0,12388497
> issuecomment-370223851
> ACJAxdQt6l0bBWLKSRRrKW3pSMdBVvmHks5tbEBzgaJpZM4SbGH4> .

I'll get ahead and add an option for disabling mill mode entirely.  I'll
also add your Winkeyer mode active code if you post it.

73
Goody
K3NG

g4fre, 2018-03-07:
Hi Anthony

 

The ability to disable mill mode would be good

 

I was working on the winkey led last night but got distracted looking at the “in function ‘void serial program memory(usb serial class) ‘ memory number may be used initialised in this function”, which led me to think in 

 

#ifdef FEATURE_COMMAND_BUTTONS

  #define analog_buttons_number_of_buttons 4

 

Does the number of buttons include or exclude the command button (ie is it the number of memory buttons?)

 

73

Dave

 

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Virus-free.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

k3ng, 2018-03-09:
> issuecomment-370254541
> ACJAxTvE6Mt2OLhccvh4J7j9RU4Cy9x5ks5tbzQtgaJpZM4SbGH4> .

analog_buttons_number_of_buttons includes the command button.  So if the
hardware has a command button and three memory buttons, set this to 4.

Uploaded updated code to Github.  There is now OPTION_EXCLUDE_MILL_MODE.

73
Goody
K3NG

k3ng, 2019-12-07:
Closing as part of issue cleanup.  73 Goody K3NG

closed 2019-12-07 by k3ng