k3ng/k3ng_cw_keyer #44: Winkey - no echo

bejoysat, 2018-03-25:
I am experiencing a problem with Winkey emulation. The keyer does not seem to send the echo and thus the client program (N1MM+ in this case) gets stuck and does not revert to RX after sending a code. 

To verify, I have tried the program WKRemote from hamcrafters.com and it connects to the keyer but responds with the error "Attach WK: No Echo"

I am using an Arduino Uno.
I have read the notes [here](https://github.com/k3ng/k3ng_cw_keyer/wiki/370-Feature:-Winkey) and tried "DisablingAutoResetOnSerialConnection" but no luck. 
I tried in Windows 10 and 7, also tried on a Nano, but the same result.

gsansoucie, 2018-03-25:
I’ve been successfully using the Winkeyer emulation for a few months now, albeit from my private fork.  I found a build time issue that has since been fixed in trunk.  My primary use is the “Iambic Master” training app and an app I wrote myself.  In my app, I did have to enable both levels of echo through the interface and it works fine.

I started with an Uno and a breadboard and made two revisions of the keyer with the Atmel chip itself.

I would not suggest building my fork as I have a conflicting function for CW send testing that isn’t documented.  It could, however, help in diffing with the current source to see if anything changed that could have affected this.

I have no experience with N1MM, is this something that “just works” if I download and install it?  I could test my configuration out for you to see if I have the same problem.

73s, de N1XF

-=Glen=-

bejoysat, 2018-03-25:
> I have no experience with N1MM, is this something that “just works” if I download and install it? I could test my configuration out for you to see if I have the same problem.

Glen, I wouldn't install N1MM for just testing this as it is quite tedious to get the software configured. You could try the original K1EL client software for PC found [here](http://k1el.tripod.com/WKremoteX.html). Its a tiny program which tests the basic functionality of the keyer.

73
de
VU3BOJ

k3ng, 2018-04-01:
What features and options do you have enabled?  Also, can you verify that
you disabled Arduino Software Reset successfully?  (The Arduino should not
reset when you load up the Arduino IDE or the Serial Monitor.)

73
Goody
K3NG

bejoysat, 2018-04-04:
> What features and options do you have enabled?

I took the original code base and just enabled FEATURE_WINKEY_EMULATION. 

> Also, can you verify that you disabled Arduino Software Reset successfully? 

Yes, I confirmed that ASR is disabled. The keyer does not reset while opening the port. I also tried enabling OPTION_WINKEY_DISCARD_BYTES_AT_STARTUP, with no luck.

73,
VU3BOJ
Bejoy

bejoysat, 2018-05-06:
> Do you have the Uno powered by the USB port? Can you try an external power supply to see if that makes a difference?
Tried but no difference. 

I tried to debug using serial port monitor to see what is going on. There is data sent to the nano, but no response. Also enabled a second debug port to print some info, but the nano doesn't seem to receive any data from the PC. I am suspecting the particular driver for the USB (CH340G) having issues with low baud rate. 

Finally, I switched over to Arduino Pro Mini and connected via a USB-TTL converter (CP2102 chip) and managed to get it work. I may stick with this now. Working with WKRemote, FLwkey and N1MM+ without any issues.

k3ng, 2018-05-08:
OK, good deal.  73

closed 2018-05-08 by k3ng