k3ng/k3ng_cw_keyer #46: winkey emulation cw speed change during sending

g4fre, 2018-03-29:
I am running the winkey emulation on a teensy 3.2.   I have speed potentiometer enabled.  Using n1mm to drive the software, if i set the cw speed to 20wpm and press the cq button in n1mm the "CQ" is sent at 20wpm the rest of the message is sent at 31wpm (the potentiometer speed). If I disable the speed potentiometer the whole "CQ" is sent at 20wpm as would be hoped. Is the potentiometer being sampled during winkey sending and over riding the speed requested by winkey?

Dave
g4fre

k3ng, 2018-03-30:
Could you post what you have defined for the N1MM function key (presumably
F1)?  Also, would it be possible to get serial port sniff log?

73
Goody
K3NG

g4fre, 2018-03-30:
Hi Goody

 

Here is the function key file

 

N1MM Logger+ CW Default Function Key File

#

# This file can be used in most CW contests that have a simple exchange

# The {EXCH} macro uses the contents of the Sent Exchange box in the contest setup

# The {SENTRSTCUT} defaults to 5NN or allows manual entry of signal report

# F5 uses "!" macro for his callsign

# Do not use the {CALL} macro in place of the ! macro

# S&P F1 calls QRL? before placing the program in RUN mode for calling CQ

#

# To respond to caller, send F5 then F2, or ; (default exchange key) or 

#   Insert, or Enter in ESM

#

#

###################

#   RUN Messages 

###################

F1 Cq,cq CQ test {MYCALL} {MYCALL} K  

F2 Exch, ! {SENTRSTCUT} {EXCH} k 

F3 Tu,! tu  QRZ {MYCALL} K 

F4 {MYCALL},{MYCALL} 

F5 His Call,!

F6 Repeat, {SENTRSTCUT} {EXCH} {EXCH}

F7 rst?, rst? 

F8 Call him, ! ! de {MYCALL} {MYCALL} k

F9 Nr?, nr?

F10 Call?,cl?

F11 LOC?, loc? 

F12 Wipe,{WIPE}

#

###################

#   S&P Messages 

###################

F1 Qrl?,qrl? de {MYCALL}

F2 Exch,{SENTRSTCUT} {EXCH}

F3 Tu,tu

F4 {MYCALL},{MYCALL}

F5 His Call,!

F6 Repeat,{SENTRSTCUT} {EXCH} {EXCH}

F7 Empty,  

F8 Agn?,agn?

F9 Nr?,nr?

F10 Call?,cl?

F11 Empty,  

F12 Wipe,{WIPE}

 

I will try and get a serial log tomorrow. I will also see what happens with my real winkey v3

 

Perhaps it’s a teensy 3.2 thing. I have another issue with it to submit tomorrow, When in standalone mode (no computer connection) hitting the memory 1 button doesn’t always send memory 1, sometimes memory 2, sometimes memory 3 same with memory 2 and memory 3 buttons. (I have 3 buttons+command) The teensy runs off 3.3V so the top end of the speed pot and the 10k pullup on the switch analog input go to 3.3V, looking at the code it uses a count not an absolute voltage for these inputs?

When I build I get:-

 

k3ng_keyer:16966: warning: "/*" within comment 

     /*

 

     ^

 

k3ng_keyer: In function 'void loop_element_lengths(float, float, int)':

k3ng_keyer:5702: warning: unused variable 'endtime' 

     unsigned long endtime = micros() + long(element_length*lengths*1000) + long(additional_time_ms*1000);

 

                   ^

 

k3ng_keyer: In function 'void service_paddle_echo()':

k3ng_keyer:11415: warning: unused variable 'character_to_send' 

   byte character_to_send = 0;

 

        ^

 

k3ng_keyer:11416: warning: unused variable 'no_space' 

   static byte no_space = 0;

 

               ^

 

k3ng_keyer: In function 'void serial_tune_command(usb_serial_class*)':

k3ng_keyer:12063: warning: variable 'incoming' set but not used 

   byte incoming;

 

        ^

 

k3ng_keyer: In function 'byte play_memory(byte)':

k3ng_keyer:14979: warning: control reaches end of non-void function 

 }

 

^

 

k3ng_keyer: In function 'void serial_program_memory(usb_serial_class*)':

k3ng_keyer:14187: warning: 'memory_number' may be used uninitialized in this function 

           EEPROM.write((memory_start(memory_number-1)+memory_index),255);  

 

                                     ^

 

Archiving built core (caching) in: C:\Users\Dave\AppData\Local\Temp\arduino_cache_905975\core\core_teensy_avr_teensy31_usb_serial,speed_96,opt_o2std,keys_en-us_71a00981eaa0bad0868b6b2ed88d327b.a

Opening Teensy Loader...

 

Sketch uses 68060 bytes (25%) of program storage space. Maximum is 262144 bytes.

Global variables use 5468 bytes (8%) of dynamic memory, leaving 60068 bytes for local variables. Maximum is 65536 bytes.

 

But these are warnings not errors

 

73

Dave

 

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Virus-free.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

k3ng, 2018-03-31:
> auth/ACJAxawkMH5P_2X3-PNXWldrxwd22KCaks5tjqyJgaJpZM4TAgTk> .

No worries with the compiler warnings....

Nothing abnormal with your F1 macro.... (I alternated lowercase and
uppercase in a test with N1MM+, no effect.)

Yea, a serial sniff is going to be a big help.

Regarding your memory button problem, this may be due to the 3.3 V
reference.  You could recompile with FEATURE_COMMAND_LINE_INTERFACE
(disable FEATURE_WINKEY) and activate DEBUG_BUTTONS.  If I recall
correctly, when you push the buttons you'll get logs showing the button
readings.  That would help us troubleshoot this.

Recently we had to add this in the settings file to accommodate a Maple
Mini which I believe has a 3.3 volt system:

#if defined(ARDUINO_MAPLE_MINI)
  #define button_value_factor 4095  //sp5iou contributed
#else
  #define button_value_factor 1023
#endif

You could try to compile with that different button value factor like so:

#if defined(ARDUINO_MAPLE_MINI)
  #define button_value_factor 4095  //sp5iou contributed
#else
  #define button_value_factor 4095
#endif

g4fre, 2018-04-01:
Changed the secret number forcing an eeprom reset

 

Don’t know if the following gives clues to the memory issue. Havent found out how to get a serial log from n1mm for the speed change issue. N1mm lets you choose in winkey settings 3 settings 1)ignore speed pot, 2)use speed pot or 3) use speed pot for paddles only. I tried all 3 but still speed changes, unless I remove the potentiometer feature

 

73

Dave

 

K3NG Keyer Version 2018.03.04.01

 

Enter \? for help

 

\S

Iambic B

Buffers: Dit On Dah On

WPM: 15

Command Mode WPM: 20

Sidetone: On 1500 hz

Dah to dit: 3.00

Weighting: 50

Serial Number: 1

Potentiometer WPM: 26 (Activated) Range: 13 - 35 WPM

Wordspace: 7

TX: 1

Quiet Paddle Interrupt: Off

Mill Mode Off

Memory 1:

 

 

\A

Iambic A

 

 

\S

Iambic A

Buffers: Dit On Dah On

WPM: 15

Command Mode WPM: 20

Sidetone: On 1500 hz

Dah to dit: 3.00

Weighting: 50

Serial Number: 1

Potentiometer WPM: 26 (Activated) Range: 13 - 35 WPM

Wordspace: 7

TX: 1

Quiet Paddle Interrupt: Off

Mill Mode Off

Memory 1:

Memory 2:{empty}

Memory 3:{empty}

Memory 4:{empty}

Memory 5:{empty}

Memory 6:{empty}

Memory 7:{empty}

Memory 8:{empty}

Memory 9:{empty}

Memory 10:{empty}

Memory 11:{empty}

Memory 12:{empty}

> 

 

 

\}1030

Setting potentiometer range to 10 - 30 WPM

 

\S

Iambic A

Buffers: Dit On Dah On

WPM: 22

Command Mode WPM: 20

Sidetone: On 1500 hz

Dah to dit: 3.00

Weighting: 50

Serial Number: 1

Potentiometer WPM: 22 (Activated) Range: 10 - 30 WPM

Wordspace: 7

TX: 1

Quiet Paddle Interrupt: Off

Mill Mode Off

Memory 1:

Memory 2:{empty}

Memory 3:{empty}

Memory 4:{empty}

Memory 5:{empty}

Memory 6:{empty}

Memory 7:{empty}

Memory 8:{empty}

Memory 9:{empty}

Memory 10:{empty}

Memory 11:{empty}

Memory 12:{empty}

> 

 

 

\P1CQ CQ CQ DE G4FRE G4FRE K

Wrote memory 1           <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<note it says memory was written

 

\P2G4FRE

Wrote memory 2

 

\P3MW2I

Wrote memory 3

 

\S

Iambic A

Buffers: Dit On Dah On

WPM: 22

Command Mode WPM: 20

Sidetone: On 1500 hz

Dah to dit: 3.00

Weighting: 50

Serial Number: 1

Potentiometer WPM: 22 (Activated) Range: 10 - 30 WPM

Wordspace: 7

TX: 1

Quiet Paddle Interrupt: Off

Mill Mode Off

Memory 1:                            <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<note memory blank

Memory 2:G4FRE

Memory 3:MW2I

Memory 4:{empty}

Memory 5:{empty}

Memory 6:{empty}

Memory 7:{empty}

Memory 8:{empty}

Memory 9:{empty}

Memory 10:{empty}

Memory 11:{empty}

Memory 12:{empty}

 

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Virus-free.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

g4fre, 2018-04-01:
I just turned on #define DEBUG_PLAY_MEMORY and logged pressing memory 1 then memory 2 then memory 3

 

I then pressed the memory 2 button 20 times and it sent memory 2 every time. With debug on I cannot get abnormal operation!

 

I then turned off debug and got the memory 2 button to send memory 3 within 4 presses!

 

73

Dave

 

From: Dave [mailto:g4fre@g4fre.com] 
Sent: 01 April 2018 22:27
To: 'k3ng/k3ng_cw_keyer' <reply@reply.github.com>
Subject: RE: [k3ng/k3ng_cw_keyer] winkey emulation cw speed change during sending (#46)

 

Changed the secret number forcing an eeprom reset

 

Don’t know if the following gives clues to the memory issue. Havent found out how to get a serial log from n1mm for the speed change issue. N1mm lets you choose in winkey settings 3 settings 1)ignore speed pot, 2)use speed pot or 3) use speed pot for paddles only. I tried all 3 but still speed changes, unless I remove the potentiometer feature

 

73

Dave

 

 

K3NG Keyer Version 2018.03.04.01

 

 

Enter \? for help

 

\S

Iambic B

Buffers: Dit On Dah On

WPM: 15

Command Mode WPM: 20

Sidetone: On 1500 hz

Dah to dit: 3.00

Weighting: 50

Serial Number: 1

Potentiometer WPM: 26 (Activated) Range: 13 - 35 WPM
[... 567 lines cut ...]
y: 220	memory_number: 2	memory_end: 299	jump_back_to_y: 9999	jump_back_to_memory_number: 255

play_memory:
    Memory number:2
    EEPROM location:221
    eeprom_byte_read:87

play_memory: reset last_memory_repeat_time
y: 221	memory_number: 2	memory_end: 299	jump_back_to_y: 9999	jump_back_to_memory_number: 255

play_memory:
    Memory number:2
    EEPROM location:222
    eeprom_byte_read:50

play_memory: reset last_memory_repeat_time
y: 222	memory_number: 2	memory_end: 299	jump_back_to_y: 9999	jump_back_to_memory_number: 255

play_memory:
    Memory number:2
    EEPROM location:223
    eeprom_byte_read:73

play_memory: reset last_memory_repeat_time
y: 223	memory_number: 2	memory_end: 299	jump_back_to_y: 9999	jump_back_to_memory_number: 255

g4fre, 2018-04-01:
Output with #define DEBUG_MEMORY_LOCATIONS

 

dave

 

\S

Iambic A

Buffers: Dit On Dah On

WPM: 22

Command Mode WPM: 20

Sidetone: On 1500 hz

Dah to dit: 3.00

Weighting: 50

Serial Number: 1

Potentiometer WPM: 26 (Activated) Range: 13 - 35 WPM

Wordspace: 7

TX: 1

Quiet Paddle Interrupt: 70

Mill Mode On

Memory 1:CQ CQ CQ DE G4FRE  start: 60 end: 139 size: 79

Memory 2:G4FRE  start: 140 end: 219 size: 79

Memory 3:MW2I  start: 220 end: 299 size: 79

Memory 4:{empty}  start: 300 end: 379 size: 79

Memory 5:{empty}  start: 380 end: 459 size: 79

Memory 6:{empty}  start: 460 end: 539 size: 79

Memory 7:{empty}  start: 540 end: 619 size: 79

Memory 8:{empty}  start: 620 end: 699 size: 79

Memory 9:{empty}  start: 700 end: 779 size: 79

Memory 10:{empty}  start: 780 end: 859 size: 79

Memory 11:{empty}  start: 860 end: 939 size: 79

Memory 12:{empty}  start: 940 end: 1019 size: 79

 

From: Dave [mailto:g4fre@g4fre.com] 
Sent: 01 April 2018 22:45
To: 'k3ng/k3ng_cw_keyer' <reply@reply.github.com>
Subject: RE: [k3ng/k3ng_cw_keyer] winkey emulation cw speed change during sending (#46)

 

I just turned on #define DEBUG_PLAY_MEMORY and logged pressing memory 1 then memory 2 then memory 3

 

I then pressed the memory 2 button 20 times and it sent memory 2 every time. With debug on I cannot get abnormal operation!

 

I then turned off debug and got the memory 2 button to send memory 3 within 4 presses!

 

73

Dave

 

From: Dave [mailto:g4fre@g4fre.com] 
Sent: 01 April 2018 22:27
To: 'k3ng/k3ng_cw_keyer' <reply@reply.github.com <mailto:reply@reply.github.com> >
Subject: RE: [k3ng/k3ng_cw_keyer] winkey emulation cw speed change during sending (#46)

 

Changed the secret number forcing an eeprom reset

 

Don’t know if the following gives clues to the memory issue. Havent found out how to get a serial log from n1mm for the speed change issue. N1mm lets you choose in winkey settings 3 settings 1)ignore speed pot, 2)use speed pot or 3) use speed pot for paddles only. I tried all 3 but still speed changes, unless I remove the potentiometer feature

 

73

Dave

 

 

K3NG Keyer Version 2018.03.04.01

 

 

Enter \? for help

 

\S

Iambic B

Buffers: Dit On Dah On

WPM: 15

Command Mode WPM: 20

Sidetone: On 1500 hz

Dah to dit: 3.00

Weighting: 50

Serial Number: 1

Potentiometer WPM: 26 (Activated) Range: 13 - 35 WPM

Wordspace: 7

TX: 1

Quiet Paddle Interrupt: Off

Mill Mode Off

Memory 1:

 

 

\A

Iambic A

 

 

\S

Iambic A

Buffers: Dit On Dah On

WPM: 15

Command Mode WPM: 20

Sidetone: On 1500 hz

Dah to dit: 3.00

Weighting: 50

Serial Number: 1

Potentiometer WPM: 26 (Activated) Range: 13 - 35 WPM

Wordspace: 7

TX: 1

Quiet Paddle Interrupt: Off

Mill Mode Off

Memory 1:

Memory 2:{empty}

Memory 3:{empty}

Memory 4:{empty}

Memory 5:{empty}

Memory 6:{empty}

Memory 7:{empty}

Memory 8:{empty}

Memory 9:{empty}

Memory 10:{empty}

Memory 11:{empty}

Memory 12:{empty}

> 

 

 

\}1030

 

Setting potentiometer range to 10 - 30 WPM

 

\S

Iambic A

Buffers: Dit On Dah On

WPM: 22

Command Mode WPM: 20

Sidetone: On 1500 hz

Dah to dit: 3.00

Weighting: 50

Serial Number: 1

Potentiometer WPM: 22 (Activated) Range: 10 - 30 WPM

Wordspace: 7

TX: 1

Quiet Paddle Interrupt: Off

Mill Mode Off

Memory 1:

Memory 2:{empty}

Memory 3:{empty}

Memory 4:{empty}

Memory 5:{empty}

Memory 6:{empty}

Memory 7:{empty}

Memory 8:{empty}

Memory 9:{empty}

Memory 10:{empty}

Memory 11:{empty}

Memory 12:{empty}

> 

 

 

\P1CQ CQ CQ DE G4FRE G4FRE K

 

Wrote memory 1           <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<note it says memory was written

 

\P2G4FRE

 

Wrote memory 2

 

\P3MW2I

 

Wrote memory 3

 

\S

Iambic A

Buffers: Dit On Dah On

WPM: 22

Command Mode WPM: 20

Sidetone: On 1500 hz

Dah to dit: 3.00

Weighting: 50

Serial Number: 1

Potentiometer WPM: 22 (Activated) Range: 10 - 30 WPM

Wordspace: 7

TX: 1

Quiet Paddle Interrupt: Off

Mill Mode Off

Memory 1:                            <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<note memory blank

Memory 2:G4FRE

Memory 3:MW2I

Memory 4:{empty}

Memory 5:{empty}

Memory 6:{empty}

Memory 7:{empty}

Memory 8:{empty}

Memory 9:{empty}

Memory 10:{empty}

Memory 11:{empty}

Memory 12:{empty}

 

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Virus-free.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

k3ng, 2018-04-02:
> auth/ACJAxZ_l7kReL4DvcyZp-AjLQtnEgIU-ks5tj40kgaJpZM4TAgTk> .

Activate DEBUG_BUTTONS and FEATURE_COMMAND_LINE_INTERFACE for
troubleshooting the buttons, not memory play debugging.

For sniffing Winkey you'll need a sniffer program.

ng7m, 2018-09-25:
I saw this problem the other day... latest code as of this note built for a nanoKeyer Rev D.  All default options.  I changed the setting in N1MM under winkeyer where I unchecked the 'pot wired with two wires' and it seemed to resolve the issue.  It was really goofy before that. Max NG7M

k3ng, 2019-12-07:
Closing this issue as part of clean up.  Please open a new issue if this is still a problem with the latest code version.  73 Goody K3NG

closed 2019-12-07 by k3ng