k3ng/k3ng_cw_keyer #48: 8 bit lcd compatibility?

dcsim0n, 2018-07-04:
Hi, I'd like to use a smaller 16x1 lcd for my keyer. The lcd I have sourced uses 8 data lines instead of 4. 
The arduino LCD library supports 8bit lcds so its just a matter of adding the data lines to K3NG code.

Anyone know which files would need modification?

Thanks
-Dana KN4BEv

k3ng, 2018-07-15:
I will add an OPTION for this.  But it's fairly easy to modify the code in
the meantime.  Change this line:

LiquidCrystal lcd(lcd_rs, lcd_enable, lcd_d4, lcd_d5, lcd_d6, lcd_d7);

to something like:

LiquidCrystal lcd(lcd_rs, lcd_enable, lcd_d4, lcd_d5, lcd_d6, lcd_d7, w, x,
y, z);

where w, x, y, and z are your four additional LCD data lines

k3ng, 2018-07-15:
Updated code on Github.

    2018.07.15.01
      Added FEATURE_LCD_8BIT for controlling standard LCD displays with 8
data lines

Please test.  Thanks

closed 2018-07-17 by k3ng

dcsim0n, 2018-07-17:
works great here, tks es 73