k3ng/k3ng_cw_keyer #58: Error detection problem in 5 char progressive training

RightRudder, 2019-02-26:
I notice that the error sidetone boop feedback doesn't come until you reach the character number of the current loop. For example if you are currently on the 3rd character but make a mistake sending the first or second character of the string there is no error detected until a 3rd character is sent on the paddles.  Shouldn't the error detection be active from the first character?  Also a pause greater than one intercharacter space should result in an error as well (IMHO).

k3ng, 2019-04-27:
This is by design and how the feature was originally written.  The code lets the user complete your sending and then informs the user if the set of characters sent is incorrect.  It repeats the current set of characters until the user sends the set back correctly.  I suppose I could make an option for immediate error feedback.  I agree on your second point about the pause, though I'd likely make that pause detection an option as well.  I may do some coding on this all soon.  Thanks and 73

VK2EFL, 2019-05-30:
Anthony,
If you are going to make a change, I agree make those extra error traps options.
I find that I know when I have a keyed an error and I just send dits to expedite the progression to the last character and then start again with that level. 73 Fred VK2EFL