k3ng/k3ng_cw_keyer #68: potentiometer_enable_pin definition missing in some hardware configurations

rificity, 2019-05-21:
I kept getting a compile error when working on a Nanokeyer Rev. D.
The line - _#define potentiometer_enable_pin 0_ - is missing from several of the _keyer_pin_settings_xxxx.h_ files.

tulipman, 2019-05-21:
Typo!!

There is a underscore missing

rificity, 2019-05-22:
No. I meant what I said. The line beginning with _#define potentiometer_enable_pin_ is missing in both of the Nanokeyer hardware pin setting configurations. I have corrected it in my local copy, but don't know how to have it updated in the repository so that others don't have to track it down.

kriskirkland, 2019-05-25:
I was able to fix this issue by adding the #define statement but see also issue #69 with the NanoKeyer rev D.

rificity, 2019-05-25:
> 
> 
> I was able to fix this issue by adding the #define statement but see also issue #69 with the NanoKeyer rev D.

As was I. My point is, the code should have that corrected so that every person that comes along with a Nanokeyer doesn't have to track it down.

kriskirkland, 2019-05-27:
> > I was able to fix this issue by adding the #define statement but see also issue #69 with the NanoKeyer rev D.
> 
> As was I. My point is, the code should have that corrected so that every person that comes along with a Nanokeyer doesn't have to track it down.

I agree, I was just commiserating with you.

k3ng, 2019-05-29:
This will be fixed shortly, version 2019.05.29.01.  Thanks and 73

closed 2019-05-29 by k3ng