k3ng/k3ng_cw_keyer #69: NanoKeyer rev D pin D2

kriskirkland, 2019-05-25:
I am noticing several strange issues, after uploading newest code upon startup the speaker "thumps" twice but does not play the normal CW HI and the D2 left side of my paddle will not work. When attempting to use winkeyer it will send these "thumps" instead of true CW. I can downgrade the code and everything works fine. I'm in the process of comparing the differences to see what the issue might be.

rificity, 2019-05-25:
Have you checked #define OPTION_SIDETONE_DIGITAL_OUTPUT_NO_SQUARE_WAVE ?

kriskirkland, 2019-05-27:
> Have you checked #define OPTION_SIDETONE_DIGITAL_OUTPUT_NO_SQUARE_WAVE ?

I have tried it both ways to no avail.

k3ng, 2019-05-29:
Please post your features and options, and keyer pin settings.

k3ng, 2019-12-07:
No activity for 6 months.  Closing issue as part of clean up.... 73 Goody

closed 2019-12-07 by k3ng