k3ng/k3ng_cw_keyer #87: 16x2 LCD won't compile

NickClark, 2019-12-12:
I'm using a LCD1602 module (non-shield). Two things of note:

1. There's no clear pin listing, just a statement that it just works. For guys like me, knowing what the pins are can be quite helpful.

2. The code doesn't compile. It complains about memory_start not being defined. (k3ng_keyer.ino
 line 7559)

For now I commented out that portion of the code and I figured out the pins (at least the screen seems to work fine). Everything seems fine, but I thought I'd file the issue here to make it better for others in the future if you can tidy it up.

Awesome project btw, thanks so much! :smile:

VK2EFL, 2019-12-18:
Nick,
The code supports quite a few different display types with different connection methods. 4 bit, 8 bit, I2C etc.
And yes the generic schematic just does not show any pins connected to a display. Perhaps because there are so many choices and perhaps because you don't necessarily require a display to use the keyer.
Assuming you are using the 4 bit connection to your LCD display then a search in the sketch shows that feature appears in only a few places. Mainly to include the necessary libraries and once to declare the variable 'lcd' and initialise it. The declaration line has a number of pins used, lcd_rs, lcd_enable, lcd_d4, lcd_d5, lcd_d6, lcd_d7, and they are defined in the file keyer_pin_settings.h.  So that is where you can find which arduino pins are in use. I know, it is a bit of a circuitous search to get the answers.

memory_start is a function. If FEATURE_MEMORIES is defined, memory_start is declared on line 16167 of version 2019.11.13.01
I don't see any listing for a call to that function near line 7559. There is a call at line 4687 and the next one down is on line 10136 (but it is commented out and so would not be included in a compile) and the next uncommented call is on line 14991.
When you attempt to compile is that error the very first one in the listing that gets produced. An error somewhere in the code can produce a rather lengthy sequence of errors and the last one to be listed is the one you see in the window where the errors get displayed, but it isn't necessarily the root cause of the problem.
Fred VK2WS

k3ng, 2020-04-24:
Hi Nick.  The memory_start error was fixed in version 2020.04.13.01.

I'll admit that documentation is lacking in areas, a lot of areas, in fact.  Sorry you had to go through some detective work, but if you'd like to contribute some documentation on this and help out the next person, I'd be happy to publish it in the wiki and give you credit, of course.

73
Goody
K3NG

k3ng, 2026-10-22:
Confirming this is still resolved on the current code - `memory_start()` is properly defined and in use on both `master` and `next_gen_version_development` (no stray/premature calls before its declaration). This was fixed back in 2020.04.13.01 per Goody's comment above; closing out this old issue.

closed 2026-10-22 by k3ng

k3ng, 2026-10-22:
Follow-up: added a short section to the top of the [Display wiki page](https://github.com/k3ng/k3ng_cw_keyer/wiki/330-Feature:-Display) pointing to external resources for physically wiring a 16x2/HD44780 LCD (Arduino's own LCD tutorial, the LiquidCrystal library reference, and SparkFun's hookup guide), since that page only ever documented which Arduino pins to tell the keyer about, not how to wire the display itself. Thanks again for flagging the gap!