k3ng/k3ng_cw_keyer #101: extended cli not working on linux (with solution)

devcpu, 2020-07-25:
All extended cli command are not working on linux because of the different lineend (\r vs. \n).
In function cli_extended_commands (line ~ 12884)

please change 
```
if (incoming_serial_byte == 13) {   // carriage return - get out
```
to 
```
if (incoming_serial_byte == 13 || incoming_serial_byte == 10) {   // carriage return - get out
```

that's it.

Can't make a pull request, but it is quite simple.

BTW. Many thanks for the software. 
73 de DL7UXA

k3ng, 2020-07-25:
What terminal program are you using?

devcpu, 2020-07-26:
The terminal in arduino ide. 
I found a lot of similar code. Some check of 10 and 13, some not. I see this because same problem programing memories from CLI.

k3ng, 2020-07-26:
Have you tried to set your Arduino IDE terminal to "Carriage Return" using
the control at the bottom of the window?

tuxun, 2020-07-26:
It think its strange cause windows end of line should be "\r\n" and the unix one "\n". https://stackoverflow.com/questions/1761051/difference-between-n-and-r

k3ng, 2020-07-26:
Fixed in version 2020.07.26.01

Thanks!
73
Goody
K3NG

closed 2020-07-26 by k3ng