k3ng/k3ng_cw_keyer #102: memory requirements for running the entire code

manotroll, 2020-08-11:
reading in some places he asks for a mega 2560 plus a much smaller esp32 r cheaper and has more memory
which cards can I run all the code on?
it seems that the nano was also indicated in another pdf

sorry for bad english i'm using translator

FrugalGuy, 2020-11-10:
Looks like a support request. Those are handled at another location.  See issue #45.

VK2EFL, 2021-05-21:
Within the Arduino family the Uno and Nano have just 32K of flash memory. You can not enable all the features and have it fit into a Uno or Nano. The Arduino Mega2560 has 256K of flash and it can fit all the features.
I don't know anything about the esp32.

manotroll, 2021-05-22:
there is no way to load the rest in some way like micro sd?
since the mega is very big

VK2EFL, 2021-05-22:
I agree, the Mega2560 is physically quite large, but it does have more flash, more RAM, more EEPROM, more interrupts and many more pins than a Uno or Nano. It is showing its age a little with no WiFi, no bluetooth, no ethernet etc.
There is no way that I am aware of to use a non motherboard flash store to hold code that gets executed.
If you want the keyer unit to act as a beacon, then it is possible to attach a SD card shield and then on that SD card store the text file that you want the beacon to transmit. But there is no way to actually store compiled code on the SD and run it.

FrugalGuy, 2021-07-10:
Can this issue be closed?

manotroll, 2021-07-10:
it may well close as it wasn't and a bug was a request

FrugalGuy, 2021-07-10:
Hi, @manotroll .  You can close by adding a comment, and instead of clicking 'Comment', click 'Close and Comment'.

closed 2021-07-10 by VK2EFL