k3ng/k3ng_cw_keyer #105: Nano Every Warning.

ve3rx, 2020-12-05:
Taking keyer code that compiles ok for a Nano, then compiling for a Nano Every gets the following warning message:

WARNING: library K3NG_PS2Keyboard claims to run on avr architecture(s) and may be incompatible with your current board which runs on megaavr architecture(s)

Need to fix the K3NG PS2 keyboard to include this megaAVR architecture.

yarda, 2023-11-04:
Does it work for you without the PS2 keyboard? For me neither, it sends periodical dots. I am afraid Nano Every port would require more work.

yarda, 2023-11-04:
Hmm, well it started to work for me with Arduino Thinary Every if I removed the FEATURE_COMMAND_LINE_INTERFACE or booted to the k3ng emulation, so ignore my previous post.

k3ng, 2026-10-22:
Fixed on both `master` and `next_gen_version_development`.

Root cause: `library.properties` for the bundled `K3NG_PS2Keyboard` library declared `architectures=avr,STM32F1`, which doesn't include `megaavr` (the architecture used by the Nano Every's ATmega4809) — that's exactly why the IDE flags it as possibly incompatible.

I checked the library's actual source and it has no classic-AVR-specific code (no raw register access — just `attachInterrupt()` and portable `avr/io.h`/`avr/interrupt.h`/`avr/pgmspace.h` includes, all present on megaavr's toolchain too), so this was a metadata gap, not a real incompatibility.

Fix: added `megaavr` to `architectures=` in `library.properties`.

Verified for real against `arduino:megaavr:nona4809` (Nano Every):
- Reproduced the exact warning you reported with the old `architectures=` list.
- Confirmed it's gone once `megaavr` is added.
- Confirmed the sketch builds cleanly either way with `FEATURE_PS2_KEYBOARD` enabled (16874 bytes flash / 634 bytes RAM on `next_gen_version_development`, 13840/589 on `master`).

@yarda — I saw your separate comment about periodical dots / deeper Nano Every issues on this thread. I wasn't able to reproduce that, and it sounded like you weren't fully sure it was a real/distinct bug either. I didn't chase it further here since it's unrelated to the warning this issue was filed for — if it's still happening, please open a new issue with the specifics (sketch config, what you're sending, what you're seeing) and I'll take a look.

Closing this one out. Thanks for the report!

closed 2026-10-22 by k3ng

yarda, 2026-10-22:
Thanks, it works for me now.