k3ng/k3ng_cw_keyer #111: CW keyer for a STRAIGHT key

glaukos78, 2021-04-07:
When connect straitght key not working.
I have try and two way with the press dah and power up and not enabled.

Also i have enable 
#define FEATURE_STRAIGHT_KEY

For you info check very old firmware and straight key work fine

VK2EFL, 2021-05-21:
Can you post a copy of the 'old version' of k3ng_keyer.ino that you think works?

glaukos78, 2021-05-21:
Hi,

Below you can find the version where working the straight key(is old)

[k3ng_keyer.zip](https://github.com/k3ng/k3ng_cw_keyer/files/6524610/k3ng_keyer.zip)

VK2EFL, 2021-05-22:
OK,
In the .ino sketch the main block of code that is used with relation to a straight key is the function service_straight_key()
I compared the code for that function from the 2.017_02_12 version to the 2021_03_20 version (the latest).
There is no difference except for the addition of three lines relating to the new feature FEATURE_LCD_BACKLIGHT_AUTO_DIM.
If you do not have FEATURE_LCD_BACKLIGHT_AUTO_DIM uncommented then it is not a factor.
So I would think that there is no change in how the straight key should work between the two versions.
There are other blocks of code that set up the pin that the straight key could be connected to, and some commands for straight key function in the CLI,  but they are the same in both versions of the code.
In the function service_straight_key() there are also blocks of code relating to FEATURE_STRAIGHT_KEY_ECHO. I wonder if you might have differences in the features_and_options.h file that might be causing different behaviour between the two versions.
I also note that in the Wiki it has -
_Go directly into straight key mode by holding down the right paddle when powering up or power resetting. This places the keyer exclusively in straight key mode with very limited functionality._

_Activate FEATURE_STRAIGHT_KEY in features_and_options.h and define pin_straight_key in keyer_pin_settings.h . Grounding the pin with a straight key will give you straight key operation in parallel with paddle operation and still give you access to the all the normal keyer functionality._

So the first way you don't need to define the FEATURE_STRAIGHT_KEY, and I think one of the paddles acts as the straight key. The second method requires you to define a new pin to be the connection to the straight key. Do you have this pin defined?

VK2EFL, 2021-05-22:
In my test setup (based on a Mega2560) I tried resetting the keyer while holding the right (dah) paddle and the keyer came up in straight key mode, where the left paddle acts like a straight key. This is as described in the Wiki.
I do not have FEATURE_STRAIGHT_KEY uncommented and I do not have a pin defined for an external straight key to connect to.
In the setup() routine is a call to check_for_beacon_mode() and here is where the state of the left paddle (to go into beacon mode) and the right paddle (to go into straight key mode) is done. This function is the same in the 2015 version of the code and the 2021 version of the code.
I don't see why this would not work for you.

VK2EFL, 2021-05-22:
The function check_for_beacon_mode(); is not executed (and straight key mode will not be set) if you have OPTION_SAVE_MEMORY_NANOKEYER uncommented. Is this what you have in your configuration?

glaukos78, 2021-06-02:
I will attached and the current configuration.

[k3ng_cw_keyer-20032021.zip](https://github.com/k3ng/k3ng_cw_keyer/files/6584852/k3ng_cw_keyer-20032021.zip)

i use the following configuration **_keyer_features_and_options_sv5fri.h_**

VK2EFL, 2021-06-02:
I have a few questions about the features and options that you have enabled and the pin settings you have specified.
Do you have a paddle connected as well as a straight key?
in the file keyer_pin_settings_sv5fri.h you have specified the straight key to be connected to pin 53. This is OK, but a pin with such a high number would only be available to you if your implementation is hosted on an ATMEGA2560 board.
In the file keyer_features_and_options_sv5fri.h you have enabled OPTION_SAVE_MEMORY_NANOKEYER. Why, when you are using an ATMEGA2560 board which has more than enough memory for all the features?
You don't say which straight key mode you are having issues with. If you have OPTION_SAVE_MEMORY_NANOKEYER enabled, then it is not possible to enter the basic straight key mode (where a paddle is used like a straight key). If you are not using an ATMEGA2560 board, and are using a Nano, then disable OPTION_SAVE_MEMORY_NANOKEYER.
The basic straight key function (where a paddle acts as a straight key) is always available when starting the keyer and holding down the right paddle (if you don't have  OPTION_SAVE_MEMORY_NANOKEYER enabled). It is not necessary to have FEATURE_STRAIGHT_KEY enabled to get this basic straight key functionality.
See the Wiki at https://github.com/k3ng/k3ng_cw_keyer/wiki/400-Operating-Modes
Other confusing options are OPTION_DISPLAY_MEMORY_CONTENTS_COMMAND_MODE when there is no display defined.
73 Fred VK2WS

FrugalGuy, 2021-07-10:
OP says he is using Mega and also says he has defined the save memory nano option set, which disables straight key. 
That’s his problem. No code issue. I think this should be closed.

closed 2021-07-10 by VK2EFL