k3ng/k3ng_cw_keyer #116: K3NG Inital speed on Arduino Nano

express405, 2021-07-25:
In my attempt to modify code so it starts at a manageable speed for learning code the NANO remains too fast for me to practice.  Build is very basic, Key, Speaker, 9v battery, Altoids tin for portability.  I do not want to have to go into command mode every time I power up.

keyer_settings_nanokeyer_rev_b.h

// Initial and hardcoded settings
#define initial_speed_wpm 18             // "factory default" keyer speed setting
#define initial_command_mode_speed_wpm 18 // "factory default" command mode speed setting 

My terminal indicates:

K3NG Keyer Version 2020.03.10.01
\S
Iambic B
Buffers: Dit On Dah On
WPM: 26
Command Mode WPM: 20
Sidetone: On 600 hz
Dah to dit: 3.00
Weighting: 50
Serial Number: 1
Wordspace: 7
TX: 1
Quiet Paddle Interrupt: Off
Mill Mode: Off
PTT Buffered Character Hold: Off
TX Inhibit: Off
TX Pause: Off
Paddle Echo: On
>

k3ng, 2021-07-26:
Do a "factory reset" by powering up with both paddles depressed.  The
should write the 18 WPM setting to EEPROM and cause the keyer to boot up at
that speed.

express405, 2021-07-27:
Worked like a charm, thank you.

closed 2021-07-27 by express405