k3ng/k3ng_cw_keyer #121: SMALL_MEM_DISPLAY MODE

F5MDY, 2021-12-15:
Hello,

If you try to put FEATURE_BUTTONS, FEATURE_COMMAND_MODE, FEATURE_MEMORIES, FEATURE_POTENTIOMETER and a LCD (FEATURE_DISPLAY), you have not enough space to enter in an arduino nano.

So I try to add a Small_Memory_Display mode in order to get a simple display mode without saturating the memory.

I selected a few useful lines in the command_mode() routine to check on the display several parameters.
And for the moment I have 95% of program storage space.

But I would want to have also an echo of the keyed characters as FEATURE_DISPLAY usually permits.

I didn't find, in the large program, where is called the procedure to print keyed characters on the display ? 

Have you an idea ? 
 
73, Nico

VK2EFL, 2022-02-25:
I think you must have some other features enabled.
I have two keyers running on a Nano clone that have Command Mode, Memories, a potentiometer and a 1602 LCD display. It fits. I am using a relatively old version of the code from 2019_12_17.
Flash 27632	89%	 RAM 1435
However the memory repeat feature does not work (hold memory button and tap paddle). I think that is because it is low on RAM. Otherwise all OK.
73 Fred VK2WS

F5MDY, 2022-03-07:
Hello Fred,
Thank you for your answer. Your 1602 LCD is not using the I2C library ??? It is connected with 4 wires ? 
73
Nico

VK2EFL, 2022-03-07:
Nico,
All my K3NG keyers that have a display (either the1602 or the 2004 type) are connected with I2C. Just 4 wires, +5V, Gnd, SDA, SCL.
As indicated, I have two keyers hosted on a Nano clone board that have Command Mode, Memories, a potentiometer and a 1602 LCD display that is connected with I2C.
Additional features, like the CLI or Winkeyer emulation, require more flash memory than the Nano has, so you must leave those features out. Here is what I have uncommented in the features and options file -
#define FEATURE_COMMAND_BUTTONS
#define FEATURE_MEMORIES                    // on the Arduino Due, you must have FEATURE_EEPROM_E24C1024 and E24C1024 EEPROM hardware in order to compile this
#define FEATURE_LCD_YDv1                                                // YourDuino I2C LCD display with old LCM 1602 V1 ic
#define FEATURE_POTENTIOMETER                                     // do not enable unless you have a potentiometer connected, otherwise noise will falsely trigger wpm changes
#define OPTION_PROG_MEM_TRIM_TRAILING_SPACES          // trim trailing spaces from memory when programming in command mode
#define OPTION_DIT_PADDLE_NO_SEND_ON_MEM_RPT          // this makes dit paddle memory interruption a little smoother
#define OPTION_DO_NOT_SAY_HI
#define OPTION_DO_NOT_SEND_UNKNOWN_CHAR_QUESTION
#define OPTION_DIRECT_PADDLE_PIN_READS_UNO  // Unos or Nanos pins 2 and 5 - do not enable on a nanoKeyer, it uses different pins
#define OPTION_PERSONALIZED_STARTUP_SCREEN     // displays a user defined string of characters on the second or fourth row of the screen during startup. 1602 display requires OPTION_DO_NOT_SAY_HI
#define OPTION_SWAP_PADDLE_PARAMETER_CHANGE_DIRECTION     // reverses the up/down direction when using the paddles to change the wpm or sidetone frequency
#define OPTION_DISPLAY_MEMORY_CONTENTS_COMMAND_MODE

73 Fred VK2WS

F5MDY, 2022-04-04:
TU Fred, effectively with these options it's working. My problem came from my display which is a OLED_SSD1306. The library should be larger and there is no enough space. I will switch my display to a more conventional 1602LCD
Best 73, Nico

VK2EFL, 2022-04-04:
Nico, I also have tried to use a small I2C 0.91inch 128x32 OLED display in an attempt to reduce the size of the enclosure required to house the finished keyer. However, like you, I found the library to support the OLED display will not fit in the available flash memory space in a Nano.
It was interesting playing with demonstration sketches to see what font size would be readable, how many lines and columns there might be and if it might be possible to have the wpm continuously displayed on one side of the display in large characters. But it didn't work out.
I then changed to using a 4 digit TM1637 LED display. I found a library that works with that display and the code does fit into a Nano. No scrolling characters as you send. But, I have a continuous display of the WPM. It required many changes to the code. And I have it all fitted into a smaller box now.
73 Fred VK2WS