k3ng/k3ng_cw_keyer #122: using the nanoKeyer

KC3TIU, 2022-01-19:
Built and working great , except for 2 issues .    1 from the start, using PS2Keyboard only F1-F3 work , no response from F4 - F12 . Then today while using keyer piezo speaker quit working properly . Will sound tone on dots , but only clicks on dashes .  Any help would be appreciated .

KC3TIU, 2022-01-19:
now making no sound at all

VK2EFL, 2022-02-25:
The nanoKeyer has its own unique configuration files. In order to save memory the nanoKeyer is constrained to only 3 memories. Hence only F1, F2 and F3 will play memories. F4-F12 are disabled.
Can't help with the piezo buzzer - sorry.
73 Fred VK2WS

g3wyw, 2022-04-03:
Guess you may have fixed this by now,but I use a nanoKEYER every day. My keyer was built from a  kit by DJ0MY and the sounder works fine. It is polarised. Did you install the correct way round by any chance?.  Not installed the PS2 keyboard support so sorry can’t comment on that problem.   - 73 Paul G3WYW

tulipman, 2023-03-16:
Some good news!I have the same nanoKeyer (DJ0MY), but unfortunately, you run out of memory during compiling the spurcecode, if you have too many options enabled. But quite recently the is a ‘turbo’ version of the nano, called ‘Nano Every’.Totally pin-compatible, but much more EPROM memory.It saved my day!

closed 2026-10-22 by k3ng