k3ng/k3ng_cw_keyer #125: QRSS mode crash - Funtronics FK-11 hardware

Olivier-ADLER, 2022-04-14:
Each time i try to switch to QRSS mode in CLI, i get a crash and reboot after the watchdog timer. If i disable the avr watchdog function, then the AVR crash and does not reboot.

This exhibit using the latest github version. Does not seem related to this particular version because the original firmware i had on the FK-11 board was a couple years younger and had the same crash problem when enabling QRSS.

Olivier-ADLER, 2022-04-15:
After checking the code it seems that the `serial_qrss_mode()` function check only the primary serial port, but on this board there is a secondary serial port. So i think that the while loop of this function never end because it never receive serial input, and trigger the AVR watchdog.

Olivier-ADLER, 2022-04-15:
Here is a working dirty code modification (dirty because it does not take into account hardware serial ports definitions).

There is still a watchdog reset problem if return is not pressed 4 seconds after the \q command. I will leave that to someone that better master the code. Other problems have been corrected.

`void serial_qrss_mode()
{
  byte looping = 1;
  byte incoming_serial_byte;
  byte numbers[4];
  byte numberindex = 0;
  String numberstring;
  byte error =0;

  while (looping) {
    if (secondary_serial_port->available() == 0) {        // wait for the next keystroke
      if (keyer_machine_mode == KEYER_NORMAL) {          // might as well do something while we're waiting
        check_paddles();
        service_dit_dah_buffers();
        //check_the_memory_buttons();
      }
    } else {

      incoming_serial_byte = secondary_serial_port->read();
      secondary_serial_port->write(incoming_serial_byte);
      if ((incoming_serial_byte > 47) && (incoming_serial_byte < 58)) {    // ascii 48-57 = "0" - "9")
        numberstring = numberstring + incoming_serial_byte;
        numbers[numberindex] = incoming_serial_byte;
//        primary_serial_port->write("numberindex:");
//        primary_serial_port->print(numberindex,DEC);
//        primary_serial_port->write("     numbers:");
//        primary_serial_port->println(numbers[numberindex],DEC);
        numberindex++;
        if (numberindex > 2)
          {
            looping = 0;
            error = 1;
          }
      } else {
        if (incoming_serial_byte == 13) {   // carriage return - get out
          looping = 0;
        } else {                 // bogus input - error out
          looping = 0;
          error = 1;
        }
      }
    }
  }

  if (error) {
    secondary_serial_port->println(F("Error..."));
    while (secondary_serial_port->available() > 0) { incoming_serial_byte = secondary_serial_port->read(); }  // clear out buffer
    return;
  } else {
    if (speed_mode != SPEED_QRSS) {
    secondary_serial_port->print(F(" Setting keyer to QRSS Mode. Dit length : "));
    } else {
    secondary_serial_port->print(F(" Changing QRSS speed. Dit length : "));
    }
    int y = 1;
    int set_dit_length = 0;
    for (int x = (numberindex - 1); x >= 0 ; x = x - 1) {
      set_dit_length = set_dit_length + ((numbers[x]-48) * y);
      y = y * 10;
    }
    if (set_dit_length == 0) {
    qrss_dit_length = initial_qrss_dit_length;
    } else {
    qrss_dit_length = set_dit_length;
    }
    speed_mode = SPEED_QRSS;
    secondary_serial_port->print(qrss_dit_length);
    secondary_serial_port->println(F(" seconds"));
    }
}`

k3ng, 2026-10-22:
Thanks for the thorough root-cause work back in 2022 - that made this straightforward to fix properly.

Confirmed: `serial_qrss_mode()` checked `primary_serial_port` unconditionally instead of whichever port the `\Q` command actually came in on. On your FK-11 (with the secondary CLI port), that left the loop waiting forever on the wrong port - nothing in it fed the watchdog timer either, so `OPTION_WATCHDOG_TIMER`'s 4-second timeout eventually force-reset the board, or it hung indefinitely with the watchdog disabled, exactly as you found. It also never echoed the digits being typed, matching your third finding.

Rather than hardcode the secondary port the way your patch did (which would just move the bug to primary-port users instead), I replaced the whole hand-rolled loop with `serial_get_number_input()` - the same helper every other numeric CLI command in this codebase already uses. It correctly uses whichever port the command came from and already echoes each digit back. Your ASCII-to-decimal conversion fix turned out to already be in place independently since 2022, so that one's covered too.

One thing I did *not* fix, flagging for visibility: the same 4-second watchdog risk exists on any numeric CLI prompt (not just `\Q`) if you pause more than 4 seconds mid-entry, since it's inherent to the shared `serial_get_number_input()` helper used throughout the CLI. Fixing that would mean touching shared code well beyond this report - a separate issue if it turns out to matter in practice.

Pushed to `master` and `next_gen_version_development`. I don't have your FK-11 hardware to confirm, so if you get a chance to flash and test `\Q` on the secondary port again, I'd be glad to hear how it goes. Closing this out - thanks again for doing most of the diagnostic legwork here!

closed 2026-10-22 by k3ng