k3ng/k3ng_cw_keyer #126: USB Keyboard - Azerty - german - french layout - how to ?

Olivier-ADLER, 2022-04-16:
I did not find any information about the method to include a different USB keyboard layout. It seems that only Qwerty US is supported.

Could someone give me the method ?

Olivier-ADLER, 2024-10-21:
Nobody knows ?

Olivier-ADLER, 2025-06-14:
Thanks but i think that this is for keyboard emulation, to output keyboard keycodes.

What i need is HID USB keyboard input with FR layout.

lengel67, 2025-06-15:
> Thanks but i think that this is for keyboard emulation, to output keyboard keycodes.

Yes indeed. Now, I see your problem but I don't have an idea on how to solve this, sorry.

Olivier-ADLER, 2025-06-17:
After looking for a solution, i've found that the keyboard layout is defined into the hidboot.cpp file, inside the USB_Host_Shield library.

closed 2026-10-22 by k3ng