k3ng/k3ng_cw_keyer #128: YAACW not connect to pc

ik6den, 2022-04-21:
Hello and sorry to me if I disturb,
i built yaacw to use it with winkeyer ( https://i1cra.briata.org/yaacwk/  )
the circuit works very well with his ps2 keyboard but does not connect to my pc. I use windows10 and CP2102 USB to UART TLL Module 6Pin Serial Converter Adapter "I also tested with
FT232RL 3.3V 5V FTDI Mini USB to TTL Serial UART Adapter Module
set serial port to 1200baud
software used is log4oM and QARtest   SO  win 7 and 10
in your opinion where am I wrong?
sorry for my english
best 73 de ik6den

ik6den, 2022-04-29:
Risolto caricando versione la precedente

federicobriata, 2022-08-22:
Hi @k3ng, can you close this issue?
The fix for have winkey working on Yaacwk was to upgrade from version 20180830 to 20190503.
thanks

closed 2022-08-22 by k3ng

ik6den, 2022-08-22:
> Hello and sorry to me if I disturb, i built yaacw to use it with winkeyer ( https://i1cra.briata.org/yaacwk/ ) the circuit works very well with his ps2 keyboard but does not connect to my pc. I use [...]

After  upgrade the software k3ng_cw_keyer, without any modification to the electrical circuit, the device works very well on both log4OM and QARTEST, thanks Federico IZ1GLG for your help 
73's ik6den