k3ng/k3ng_cw_keyer #129: Can't connect the paddle to pin-IO0?

jh4vaj, 2022-05-19:
My hardware has the left paddle connected to pin-IO1 and the right paddle connected to pin-IO0.
And it works fine with the 2021.12.17.01 software.

Yesterday, I downloaded the new source code (2022.01.28.01) and compiled it, but I got a compilation error.
When I checked the history of the source code, it was as follows.

> Added compiler macro to error out if paddle_left or paddle_right is defined as 0 (disabled)

I have keyer_pin_settings.h like this to fit my hardware:

```
#define paddle_left 1
#define paddle_right 0
```

I thought that this "0" was not the meaning of disable, but the definition of the pin number (IO0).

Is it no longer possible to assign a pin-IO0 to a paddle with current software?

jefflogullo, 2022-05-19:
Hello JH4VAJ-san,

If I’m reading this right... this was the result of a long troubleshooting session another user had. You should reference the thread on groups.io:

#16471  <https://groups.io/g/radioartisan/message/16471> is the analysis by Fred VK2WS which got to the bottom of the issue - a user was defining paddle pins with ‘0’ to disable (he wasn’t connecting paddles; only using the keyer as a Winkeyer).

Also see the resulting change committed on 28 Jan:

https://github.com/k3ng/k3ng_cw_keyer/commit/8234510277de7437973971e2997b43a1fe64f614 <https://github.com/k3ng/k3ng_cw_keyer/commit/8234510277de7437973971e2997b43a1fe64f614>

#if (paddle_left == 0) || (paddle_right == 0)
  #error "You cannot define paddle_left or paddle_right as 0 to disable"
#endif

You may have to edit the source to re-enable the use of pin 0 as a paddle input. Or (if possible) use a different pin. That’s a one-time hardware change as a trade-off for always having to modify source when updating your code.

Hope this helps!

73,
Jeff N0̸MII

> On May 19, 2022, at 4:20 AM, jh4vaj ***@***.***> wrote:
> 
> 
> My hardware has the left paddle connected to pin-IO1 and the right paddle connected to pin-IO0.
> And it works fine with the 2021.12.17.01 software.
> 
> Yesterday, I downloaded the new source code (2022.01.28.01) and compiled it, but I got a compilation error.
> When I checked the history of the source code, it was as follows.
> 
> Added compiler macro to error out if paddle_left or paddle_right is defined as 0 (disabled)
> 
> I have keyer_pin_settings.h like this to fit my hardware:
> 
> #define paddle_left 1
> #define paddle_right 0
> I thought that this "0" was not the meaning of disable, but the definition of the pin number (IO0).
> 
> Is it no longer possible to assign a pin-IO0 to a paddle with current software?
>

jh4vaj, 2022-05-20:
Thank you very much, Jeff. 

I understand the situation.

I have just completed my variant.
It has been working fine with the previous software (paddles are defined on pin-IO0 and pin-IO1).
I ran into this trouble when I tried to replace it with the latest software.

I understand that changing the pin assignments is a good idea, but doing that now is difficult.
Because it would require a redesign of the PCB.

I compared the previous software with the latest one and it seems that the following code has just been added.

```
#if (paddle_left == 0) || (paddle_right == 0)
  #error "You cannot define paddle_left or paddle_right as 0 to disable"
#endif
```

This time, this code will be disabled.
If someday I ever redesign the PCB, then I will change the paddle pin definitions.

If this trouble is limited to cases with CLI functionality, how about the following code?

`#if defined(FEATURE_COMMAND_LINE_INTERFACE) && ((paddle_left == 0) || (paddle_right == 0))`

Mit / JH4VAJ

closed 2026-10-22 by k3ng