k3ng/k3ng_cw_keyer #130: I2C LCD adaptor module chip PCF8574T

iv3ynb, 2022-05-23:
Hi, 
I am going to build a cw keyer and I wish to use I2C display to save pins.
I have an I2C lcd adaptor module with PCF8574T chip on it, may I use it? Should I uncomment the following line?

**// #define FEATURE_LCD_ADAFRUIT_I2C    // Adafruit I2C LCD display using MCP23017 at addr 0x20**

I think the answer will be "no, you cannot", anyway I ask to see if there is some compatibility between chips that I am missing...

Thanks for helping hand
Matteo IV3YNB

tulipman, 2022-05-23:
Check this out:

https://groups.io/g/radioartisan/topic/68630322?p=Created,,,20,2,0,0

iv3ynb, 2022-05-23:
Thank you so much, I think this will solve the trouble.

Matteo IV3YNB

iv3ynb, 2022-05-23:
Thanks for the infos: I want to use an Arduino MEGA 2560 for the reasons you talked about.
I'm not so much into Arduino so far, because I use to deal with Microchip PICs in MPLABX environment using C language.

I'll surely take a look at those new Arduino boards, simply for they'll have debug option in IDE 2.0: a feature that I am really missing in the 1.0 IDE, since I come from MPLABX environment.

73 de IV3YNB, 
one that would "run" on CW but... it's soooo hard ;-)

tulipman, 2022-05-23:
> 73 de Matteo IV3YNB,

Well, I hated the standard Arduino IDE from the start.
I’m now using the Microsoft visual Studio Code.

Or, if you are not really a MS fanboy, have a look at the ATOM editor.

I also used the (not free) Sublime text editor, which is amazingl!

Would be cool if you could send me a video of your working K3NG keyer when
it’s up and running!!

BTW: I’m using Begali paddles!!
Have the Stealth Limited (nr 33), and also the Graciella (a piece if
Italian art!!)

I’m currently QRV using my wonderful ICOM IC-705 using a Bufdipole DeLuxe.
Awsome combination!!

Ciao
73 de Robbert / PA3BKL (aka TA2IX)

iv3ynb, 2022-05-23:
> Well, I hated the standard Arduino IDE from the start. I’m now using the Microsoft visual Studio Code. Or, if you are not really a MS fanboy, have a look at the ATOM editor. I also used the (not free) Sublime text editor, which is amazingl!  <

Cool... I hated it too since I bought my first Arduino Uno when it came first out on the market, so I left... I just got into Arduino once more because of in my new job they use to program some boards...  Do these editors you talk about, feature "line by line" debugging and/or simulation/stimulus???

> Would be cool if you could send me a video of your working K3NG keyer when it’s up and running!! <

Yes of course, I only have limited time to build it... please have patience :-D 

I've know Begali paddles, but I admire also your "Morsetasten" ;-)

73
Matteo IV3YNB

tulipman, 2022-05-23:
> Well, I hated the standard Arduino IDE from the start. I’m now using the Microsoft visual Studio Code. Or, if you are not really a MS fanboy, have a look at the ATOM editor. I also used the (not [...]
It does!!
Just set breakpoints and go debugging:
Check this out:

https://youtu.be/6cOsxaNC06c

tulipman, 2022-05-23:
So you checked out my website?

Sent from my iPhone XS Max (512Gb)

> On 23 May 2022, at 13:41, iv3ynb ***@***.***> wrote:
> 
> I've know Begali paddles, but I admire also your "Morsetasten" ;-)

iv3ynb, 2022-05-23:
> It does!!
Just set breakpoints and go debugging: <

OK, now I understand how it is possible to write long listings for Arduino boards, without entering hundreds of "serial print" commands !!!

> So you checked out my website? <

Yes ! I always search for hams websites, usually starting from QRZ : it's very interesting to see "who's who" ;-) 

73
Matteo IV3YNB

tulipman, 2022-06-05:
Hi Matteo !

Just wondered how your cw keyer project is going.

Are you winning??

73 de Robbert / PA3BKL aka TA2IX

iv3ynb, 2022-06-06:
I am building it... I used an Arduino Mega 250 R3 connected to a 4x20 LCD via I2C and 6 buttons for command mode and macros.
I am now focusing on WPM adjustment via pot. This because I have a pot "here and now", but I planned to do it with a rotary encoder that I like most... gotta find it my various junk boxes and drawers... 
A friend of mine gave me a sidesweeper key to try creating code, but it's quite slow and not reliable due to switch reed contacts: it is a 3D printed project realized for a QRS beginner course. Not good for operating as a Graciella is ;-) but it works to get dits and dats.

I wish to get WPM indications on two 7 segment led displays but I cannot recognize the variable in the code to manage and visualize on the displays... can you help me this way?

I'll take some pictures asap.

tulipman, 2022-06-06:
Hi Matteo!!

Yes, rotary encoders are absolutely needed!
Don’t go for the standaard potmeter type.

I’m now investigating on adding a graphic touch display.
That will be such a delight to use, especially macro functions.

And stay far from paddles with microswitches.
Just use gold- or silverplated contacts.

In 1981 I found a el-bug design in the ELEKTOR magazine.
It used microswitches, and it was really un-usable.
I learned a LOT from it.

My first paddle was a Bencher BY-01.
I loved it, and still use it.

If u need idea’s or help, let me know !!

73 de Robbert  / TA2/PA3BKL from Istanbul

tulipman, 2022-06-06:
WPM indication is not so hard to do.
There IS a WPM variable available in the code.

Did you see the fantastic K3NG keyer that was made by your compatriot IZ3ZLU ?

Check this out:
Using a NEXTION touch display

https://youtu.be/bkGq9O5qkUQ

iv3ynb, 2022-10-11:
Hi Robbert,
I am building it... I used an Arduino Mega 250 R3 connected to a 4x20 LCD via I2C and 6 buttons for command mode and macros.I am now focusing on WPM adjustment via potentiometer. This because I have a potentiometer "here and now", but I am planned to do it with a rotary encoder that I like most... gotta find it my various junk boxes and drawers... :-DDD
A friend of mine gave me a sidesweeper to try creating code, but it's quite slow and not reliable: it is a 3D printed project realized for a QRS begginner course. Not good for operating as a Graciella is ;-) but it works to get dits and dats.

I wish to get WPM indications on two 7 segment led displays but I cannot recognize the variable in the code to manage and visualize on the displays... can you help me this way?

I'll take some pictures for you, asap.73, Matteo

Matteo Copetti -- IV3YNB -- JN65urhttp://iv3ynb.altervista.org

closed 2026-10-22 by k3ng