k3ng/k3ng_cw_keyer #137: Compillation error for Generic STM32F103C

Sergey-R5AU, 2023-02-19:
Hi !
I have error with compillation keyer code for STM32 Blue Pill board:
==========================================
C:\Users\r5au\Documents\Arduino\k3ng_cw_keyer-master\k3ng_keyer\k3ng_keyer.ino:1536:4: error: #error "You cannot define paddle_left or paddle_right as 0 to disable"
   #error "You cannot define paddle_left or paddle_right as 0 to disable"
    ^
Multiple libraries were found for "Wire.h"
  Used: C:\Users\r5au\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2022.9.26\libraries\Wire
  Not used: C:\Users\r5au\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2022.9.26\libraries\WireSlave
exit status 1

Compilation error: #error "You cannot define paddle_left or paddle_right as 0 to disable"
===========================================

Arduino IDE version 2.0.3 64bit

STM32 board support with http://dan.drown.org/stm32duino/package_STM32duino_index.json

What is wrong ?
Copmpillation for Arduino nano without troubles

k3ng, 2023-02-19:
You have paddle_left and/or paddle_right defined as pin 0. Assign real pins
to these two.

Sergey-R5AU, 2023-02-19:
Hi Antony !
Thank you for reply !
in my definitions: keyer_pin_settings_generic_STM32F103C.h looks like:
---------------------------------------------------------------
GENERIC STM32F103C Blue Pill board                                                                                  

*/

/* Pins - you must review these and configure ! */
#ifndef keyer_pin_settings_h
#define keyer_pin_settings_h

#define **paddle_left PA1**
#define **paddle_right PA0**
#define tx_key_line_1 PC15       // (high = key down/tx on)
.
.
--------------------------------------------------------------
or i should directly wrote here pins nr from Blue Pill, like:
#define **paddle_left 11
#define **paddle_right 11
following with attached pic ?

[image]

Sergey-R5AU, 2023-02-19:
For tetsing purpose i did as abose 
#define paddle_left 11
#define paddle_right 10

however next stop with Serial Port:
----------------------------------------------
C:\Users\r5au\Documents\Arduino\k3ng_cw_keyer-master\k3ng_keyer\k3ng_keyer.ino: In function 'void initialize_serial_ports()':
C:\Users\r5au\Documents\Arduino\k3ng_cw_keyer-master\k3ng_keyer\k3ng_keyer.ino:18238:25: error: cannot convert 'HardwareSerial' to 'USBSerial' in assignment
     primary_serial_port = PRIMARY_SERIAL_PORT;
     
Compilation error: cannot convert 'HardwareSerial' to 'USBSerial' in assignment
----------------------------------------------
probably somwhere in definitions files port assigment are wrong ?  'HardwareSerial' <-> 'USBSerial'  ?

pe1cim, 2023-03-14:
Do you have the correct board loaded ?
Load the stm boards on IDE. 
Preferences -> Board Manegers URL
Use :https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json

k3ng, 2026-10-21:
Fixed — thanks @pe1cim for the pointer to the STMicroelectronics board package, that confirmed the real cause.

The errors in this thread (the paddle_left/paddle_right == 0 error, then the cannot-convert-HardwareSerial-to-USBSerial error) were both downstream symptoms, not the actual bug. This hardware profile was written against the older, now largely unmaintained "Arduino_STM32" (roger-clark) core, which defines `ARDUINO_GENERIC_STM32F103C`/`ARDUINO_MAPLE_MINI` directly. The modern, actively-maintained core most people install today (`STMicroelectronics:stm32`, the one @pe1cim suggested) instead defines a distinct macro per board selection — `ARDUINO_BLUEPILL_F103C8`, `ARDUINO_GENERIC_F103C8TX`, `ARDUINO_BLACKPILL_F103C8`, etc. — none of which matched, so the sketch fell all the way through to the AVR-only fallback and failed immediately on `avr/wdt.h`, before ever reaching the paddle-pin or serial-type code you both got stuck on.

Fixed:
- Added recognition for the modern core's Blue Pill/Black Pill/generic F103C6/C8/CB board selections.
- `PRIMARY_SERIAL_CLS` was hardcoded to `USBSerial`, but the modern core's "USB support" board option defaults to "None" — plain `HardwareSerial` on `Serial`. `USBSerial` only applies if you explicitly select USB CDC mode in the Tools menu, which is exactly the conversion error you both hit. Now defaults to `HardwareSerial` to match the out-of-the-box board config, with a new `OPTION_STM32_USB_CDC_SERIAL` opt-in if you have enabled USB CDC mode.
- Skipped a couple of legacy-core-only compatibility shims (a forward-declaration workaround, some dead `EEPROM.init()`/`format()` calls) that don't apply to the modern core.
- A couple of unrelated `char`→`String` portability bugs turned up while testing the Maple Mini profile and got fixed too.

Pushed to `master` and `next_gen_version_development`. Build-tested for real against `STMicroelectronics:stm32` 3.0.0 (not just a syntax check): `BLUEPILL_F103CB`, `GENERIC_F103CBTX`, and `MAPLEMINI_F103CB` all compile clean. One caveat: the classic 64KB Blue Pill (`F103C8`) compiles correctly but doesn't fit this project's full default feature set in flash (~17.6KB over) — that's a capacity trade-off specific to that smaller chip, not a bug, so if you're on an F103C8 you may need to disable some features to fit. The 128KB variants (`F103CB`, sold as "BluePill F103C8 (or C8 with 128k)" in some board lists) have plenty of room.

Closing this out — thanks for the detailed reports, and let us know if you run into anything else getting it running.

closed 2026-10-21 by k3ng