k3ng/k3ng_cw_keyer #138: Issue with adding keyboard library K3NG_Keyboard in latest version of Arduino IDE

m0nde, 2023-02-26:
I am unable to add the K3NG_Keyboard library using the latest IDE.  It seems the format of the library may need updating to conform with the library formats required by the IDE.  I have built a working keyer before but this was perhaps 2 years ago.  The IDE does not see the library files.
  Is there a work around please?

jefflogullo, 2023-02-27:
Did you uncomment the three USB-related lines (look around lines 1631) in k3ng_keyer.ino?
Search for "#if defined(FEATURE_USB_KEYBOARD)"...
 If they are commented then you'll see  lots of USB-related errors when compiling.
I just verified that I can compile for an FK-10 (Mega2560) that uses USB keyboard with Arduino IDE 2.0.3.
If that doesn't help can you give more details? LIke: path to your sketchbook; where your Libraries folder resides; etc.

m0nde, 2023-02-27:
Solved with the help of Jeff N0MII.  The LCD library needed to reside in the IDE program library directory not the keyer code directory.

closed 2023-02-27 by m0nde