k3ng/k3ng_cw_keyer #142: Would this work as only a WinKeyer emulator?

kj7rrv, 2023-11-05:
I am interested in creating a Winkeyer-compatible keyer that only keys based on computer input. For my use case, I don't need to be able to connect paddles or a keyboard to the keyer; I just need to put computer commands in using the Winkeyer protocol and get simulated straight key output out. Is that possible to do with this firmware?

kj7rrv, 2023-11-05:
It looks like nanoIO is designed more for what I want than this is, so I think I'll go with that.

closed 2023-11-05 by kj7rrv

k3ng, 2023-11-05:
A lot of people use this keyer with just Winkey functionality driving a transmitter.

jefflogullo, 2023-11-05:
See Mortty:https://hamprojects.info/morttyK3NG software works perfectly for the WinKeyer emulation use case!Jeff N0̸MII