k3ng/k3ng_cw_keyer #144: Question: Arduino Pro Micro timing issue - akt's like as in timelaps after powercycle

klaute, 2024-04-28:
Hello,

I'm currently try to get the Keyer firmware to work.

Arduino IDE 2.3.2, and a Arduino Pro Micro board attached to a Mac M1 (latest OS).

I connected a 3D printed CW key to the pins 2 and 5, and also a small speaker to pin 4.

I can hear the dit's and dah's if I press the keys. Also I can see the letters written by the HID keyboard of the firmware and also the letters on the serial monitor in the Arduino IDE.

All I have enabled in the keyer_features_and_options.h:
```
#define FEATURE_BUTTONS
#define FEATURE_COMMAND_LINE_INTERFACE
#define FEATURE_CW_COMPUTER_KEYBOARD
```

### Issue i'm facing:

After I reconnect (powercycle) the Arduino, the speed of the audio is very slow. Pressing the dah-key cause a low frequency beep for about a second. A dit-key about half of a second.

I filmed it to show the behavior...

After flashing:
https://share.icloud.com/photos/04b768W80qP0NPxG7pj9ogWAA

Afte USB reconnect (power cycle):
https://share.icloud.com/photos/046mfH6vW-iJAYGX1sYE1B36g

Serial monitor output:
```
K3NG Keyer Version 2024.03.20.2239

...some key output...
```

Also on the serial interface the letters are received as slow as the audio output. Same for the USB HID keyboard letters.

Did I missed something to configure?

73
DO4KLA

klaute, 2024-04-29:
I don't know the reason. But a workaround is to reset the Arduino manually.
This fixed the timing issue.

k3ng, 2026-10-21:
Thanks for the detailed report, and for the workaround you found (manual reset) - that's actually a strong clue.

There's no clock-prescaler handling anywhere in this sketch, which points to a known class of issue on some ATmega32U4 clone boards (Pro Micro, Leonardo, Micro, etc.): the bootloader only correctly sets the CPU clock prescaler during the IDE's upload sequence (reset-into-bootloader), not on a genuine cold power-on. That can leave the MCU running at the factory-default safety prescaler - a fraction of its intended speed - until a manual reset re-triggers correct initialization. Since every `millis()`/`delay()`-derived timing in this sketch (tone duration, CLI/HID character pacing, CW element timing) assumes the real clock speed, that would explain the uniform slowdown you saw across audio, serial, and USB HID output alike, and why a reset fixes it.

Added `OPTION_FIX_SLOW_CLOCK_AFTER_POWER_CYCLE` (just pushed to `master` and `next_gen_version_development`, documented in the wiki's Features and Options page). Enable it in `keyer_features_and_options.h` and it'll force the clock prescaler to full speed as the very first thing in `setup()` - it's a no-op (and therefore safe to leave on) if your board doesn't actually have this issue.

I don't have your specific clone board here to confirm this resolves it, so if you get a chance to flash it and try, I'd be glad to hear how it goes. Closing this out for now since it's been quiet a while - feel free to reopen if the option doesn't fix it for you.

closed 2026-10-21 by k3ng