k3ng/k3ng_cw_keyer #145: Bug Mode

QueenCityNerd, 2024-07-24:
So I have a TenTec Rebel radio and part of the keyer that the rebel alliance mods included was based on the open QRP keyer. 

  What I'm trying to do is make my iambic into a bug mode and searching the code change that will allow me to understand how this is done. 

  Where is it in all this, I've not found anything commented Bug except that it's option g 

  Thanks, and 73, 

 N2QFD//Mal

nshin31302, 2025-01-01:
This is a late response, but in the Arduino serial monitor or putty, you can type \g , and that will set it into bug mode. I believe the keyer will "remember" what mode it is in even after power is removed. If you want to reset it to iambic mode a or b you can use \a or \b

k3ng, 2026-10-21:
Thanks @nshin31302 — that's correct. \G (Bug mode) and \A \B (Iambic A/B) are CLI commands — type them into a serial terminal connected to the keyer
  (Arduino Serial Monitor, PuTTY, etc.), and yes, the mode is saved and will persist across power cycles.
             
@QueenCityNerd closing this out as answered — let us know if you run into anything else getting it working on the Rebel.

closed 2026-10-21 by k3ng