k3ng/k3ng_cw_keyer #148: Pressing the command button occasionally corrupts the memory1 slot

lagomorph, 2025-01-09:
When this happens I only get the first tone of two. I'm using a STM32 Blue Pill.

k3ng, 2025-01-09:
Can you post your compiler executable size / local variable space available messages?

lagomorph, 2025-01-09:
Sketch uses 52112 bytes (79%) of program storage space. Maximum is 65536 bytes.
Global variables use 2896 bytes (14%) of dynamic memory, leaving 17584 bytes for local variables. Maximum is 20480 bytes.

Opening and parsing file: k3ng_keyer.ino.bin
  File          : k3ng_keyer.ino.bin
  Size          : 51.19 KB 
  Address       : 0x08000000

k3ng, 2026-10-21:
Hello,
  
  First investigation found the button-selection logic wasn't properly averaging its 19 ADC samples, so noisy readings could occasionally resolve to the
  wrong button — matching your original report. That fix led to a regression during hardware testing (every button resolving to the highest-numbered one),
  traced to mechanical contact bounce: 19 back-to-back analogRead() calls finish in under 1ms on AVR, well before a real button's contact has finished
  settling (which can take several ms to tens of ms). Fixed by spreading those reads out over a couple of milliseconds each so they land after the contact
  has actually settled, and by switching to a proper per-sample majority vote instead of averaging raw voltages (which was itself skewed by the bounce
  transient).
  
  Verified working correctly on hardware. Pushed to master and next_gen_version_development. Thanks for the detailed report and for bearing with the
  back-and-forth — if you get a chance to confirm on your STM32 Blue Pill too, that'd be great, but I think this is good to close.

lagomorph, 2026-10-22:
Thank you for the detailed description. With this fix I haven't been able to reproduce the memory corruption issue and the other memory buttons seem to work reliably. The other buttons were also iffy at times before. My command button doesn't enter command mode but I may be missing something in the configuration. I have  FEATURE_BUTTONS and FEATURE_COMMAND_MODE defined. I'll keep looking at it. Thanks again.

k3ng, 2026-10-23:
Great!  Apologies for the long response time for this issue.  I recently adopted Claude Code and it's now doing all the heavy lifting for me. :-). 

If you experience any problems with the updated code, please activate DEBUG_BUTTONS, copy the debug logs when pressing various buttons, and re-open this issue and post the logs here.

Thanks!
73
Goody
K3NG

lagomorph, 2026-10-23:
When I press the command button the pin reads 0. If I'm reading the code right that looks the same as no button pressed. check_buttons() has code to enable command mode but I don't see how it would be reached. Am I missing where the command button is handled in a special way? 

This appears to be a workaround for me.

```
diff --git a/k3ng_keyer/src/buttonarray/buttonarray.cpp b/k3ng_keyer/src/buttonarray/buttonarray.cpp
index 8079d38..9bc2c2a 100644
--- a/k3ng_keyer/src/buttonarray/buttonarray.cpp
+++ b/k3ng_keyer/src/buttonarray/buttonarray.cpp
@@ -34,7 +34,7 @@ void Button::InitLimits(uint8_t step, int32_t low_limit, int32_t high_limit){
 }
 
 bool Button::Pressed(int32_t analog_reading){
-    if ((analog_reading > low_limit_) && (analog_reading <=  high_limit_)) {
+    if ((analog_reading >= low_limit_) && (analog_reading < high_limit_)) {
         return true;
     } 
     return false;
```

k3ng, 2026-10-23:
Great find, thank you! Confirmed - this is a real bug, and a genuinely independent one from the mechanical-bounce/averaging issue fixed earlier in this thread.

Root cause: `Button::InitLimits(step)` computes each button's `low_limit_` by averaging against a "virtual neighbor" at `step-1`. For button 0 (the command button - the lowest step, with no real neighbor below it), that makes `step-1` negative, so the inner expression legitimately comes out negative too - this function's own doc comment even documents the expected result ("0: -56 - 46"). But the three intermediate values were cast to `uint32_t`, and casting a negative float to an unsigned type is undefined behavior in C++ - in practice this collapsed `low_limit_` for button 0 to exactly `0` on your toolchain. `Button::Pressed()`'s low-bound check is a strict `>`, so a button-0 press reading exactly `0` (very plausible for the lowest step, often a dead short to ground) failed to match its own button - looking identical to "no button pressed," exactly matching what you found.

Your proposed workaround (flipping the comparison to `>=`/`<`) does mask this specific symptom, but only because the broken value happens to land on exactly `0` on your platform - that's not guaranteed on every architecture, since the underlying cast is still undefined behavior. I fixed the actual cast instead (`int32_t` rather than `uint32_t`), which restores the negative `low_limit_` the code's own documentation always said button 0 should have, and doesn't change anything for any other button.

Pushed to both `master` and `next_gen_version_development`. I don't have hardware to verify this on, so if you get a chance to confirm the command button now works reliably (especially on your STM32), that'd be great - and if anything's still off, DEBUG_BUTTONS should now show you the raw reading alongside the resolved button index to help narrow it down further.

Thanks again for digging into this!

73
Goody
K3NG

lagomorph, 2026-10-23:
Makes perfect sense, thank you. This fix works fine on my STM32. Thanks again!

k3ng, 2026-10-23:
Excellent!  Claude wrote all of that, but this text here is from Goody, the human.

:-)

73
Goody
K3NG

closed 2026-10-23 by k3ng