k3ng/k3ng_cw_keyer #151: Add support for hd44780 pinIO module

julia-m-stephenson, 2025-06-22:
I've been trying to get Old Hitachi 4x16 display modules (LM041L) with genuine HItachi  HD44780 controllers to work with K3NG Keyer. 

I found using the pinIO module with the HD44780 driver worked because it supports slowing down the write accesses with  function lcd.setExecTimes

I will prettify my mods and create a pull request. Hopefully this is okay? I was a embedded software engineer until I retired recently!

julia-m-stephenson, 2025-06-22:
Please assign to me.

k3ng, 2026-10-21:
This was addressed in #152 (merged) — thanks for following through with the PR! `FEATURE_LCD_HD44780_PINIO` with `setExecTimes()` tuning is in the codebase now. Closing this out as resolved.

closed 2026-10-21 by k3ng